supermerill/SuperSlicer #1240: Ability to set up paths for Filament settings, print settings, and printers.

rjmessing, 2021-06-04:
I'd like to use multiple computers and use a network shared SuperSlicer load.  I'd like to use my profiles for filaments, printer and print settings across all machines, instead of the computer specific, local cache under appdata/roaming/superslicer.  Having the ability to configure the directories to something other than a local cache would help, if SuperSlicer is used on a network share.  Another thought is to have the profiles are saved into the SuperSlicer directory and not as a local cache.

neophyl, 2021-06-04:
You can already do this.  At least on windows machines.  When you launch the software you can define the datadir with a switch.  So you just use a .bat file and use that to launch.  

In fact in the folder with the Super Slicer.exe there is already a local settings batch file which if ran from the folder will start it and create a settings folder in that directory and then start it as a clean install as there's no profiles already.

For example a start.bat file containing the following
".\superslicer.exe --datadir f:\profiles" would start Super slicer using the profiles contained in f:\profiles

The normal place the software stores its profiles in can be found through going to Help>Show configuration folder.  You can copy that folder anywhere, say Dropbox for example and then point the batch file at the new folder location.

Then when launched from the batch file (or amend the shortcut link in windows directly) it will use that new location.  Its how I used to run it on multiple computers with the profiles folder stored in Dropbox so it was always synced.

I don't bother with that any more though.  I don't even install it anymore, I run it from a USB stick.  I have my profiles folder on the stick along with multiple versions of the software and I just plug it into any machine I'm working on and can start it and use anywhere now.

rjmessing, 2021-06-05:
@neophyl thanks for your help

closed 2021-06-05 by rjmessing