supermerill/SuperSlicer #1241: UI shaking before and after slice action

uocnb, 2021-06-04:
**Describe the bug**
Sidebar and main view UI are jumping / shaking before and after slice due to `Sliced info` section only visible after sliced. 

**To Reproduce**
Steps to reproduce the behavior:
1. Import model
2. Select model
3. Add object settings to fill the empty space above `Info` section and make sure no scrollbar show yet.
4. Click on Slice button
5. Scrollbar show because of additional section `Sliced info` just added to the sidebar caused the sidebar shaking.
6. 3D Plater tab switched to GCode preview tab caused UI shaking.

**Expected behavior**
Consistence UX without jumping / shaking before and after slice action.

**Screenshots**
Before slice: 
[image]

After slice:
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 2.3.56.1

supermerill, 2021-06-04:
can you try with the latest nightly? I fixed a "shaking" but I don't know if it's that.

uocnb, 2021-06-04:
Cool, I will try it tonight.

supermerill, 2021-06-04:
can't reproduce with my dev build, or I miss something.

uocnb, 2021-06-04:
Just checked, I'm using the latest build downloaded from git release page already.

supermerill, 2021-06-04:
https://github.com/supermerill/SuperSlicer/suites/2905660844/artifacts/65159859
from https://github.com/supermerill/SuperSlicer/actions

uocnb, 2021-06-04:
Thank @supermerill! It's definitely better with the nightly build!

closed 2021-06-13 by supermerill