supermerill/SuperSlicer #1242: Gaps in first layer

G-r-i-s-h-U, 2021-06-04:
**Describe the bug**
This version (unlike some previous) leaves unfilled gaps in first layer.

**To Reproduce**
Slice a part with holes around the perimeters (or sharp edges).

**Expected behavior**
Expect to fill them all, not just partly.

**Screenshots**
Attaching screenshots from current version (2.3.56.0) and a previous one (2.3.55.2) (both using same profiles).

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 2.3.56.0

[image]
[image]

supermerill, 2021-06-04:
I need your project file

G-r-i-s-h-U, 2021-06-04:
Here it is. (I have archived it, since it doesn't want to upload as *.3mf)
[Case_Cap.zip](https://github.com/supermerill/SuperSlicer/files/6599526/Case_Cap.zip)

supermerill, 2021-06-04:
Same issue as #1214, i'll close this one.

closed 2021-06-04 by supermerill