supermerill/SuperSlicer #1244: Internal error: invalid string position

simenon, 2021-06-04:
### Version
2.3.56

### Operating system type + version
Windows

### Behavior
Superslicer crashes at startup 
[image]

supermerill, 2021-06-04:
can you zip and post here your setting directory (should be c:/user/yourname/appdata/roaming/superslicer) ?

DKB234, 2021-06-05:
I had that same problem as 'simeon' and to fix it I renamed this file ' c:/user/yourname/appdata/roaming/superslicer.ini' and it now loads. I renamed the file just in case I was wrong, bu all good for me. 
[image]
If this image is vewable you can see the difference. Left one works, right side caused the crash.

supermerill, 2021-06-05:
should be fixed in nighty

simenon, 2021-06-05:
[SuperSlicer.zip](https://github.com/supermerill/SuperSlicer/files/6602427/SuperSlicer.zip)
Adjusting only the ini file does not work for me. Included my settings directory

closed 2021-06-13 by supermerill