supermerill/SuperSlicer #1245: 2.3.56 - Bed on thumbnail uncheck still generate bed on thumbnail

excessnet, 2021-06-04:
**Describe the bug**
If you uncheck the box "Bed on thumbnail", the bed is still shown on the thumbnail on Mainsail.

[image]

Up : with 2.3.56.2
Buttom : with 2.3.55.x
[image]

millerlarson, 2021-06-08:
For what it's worth, I just upgraded to 2.3.56.3 and I'm still seeing the bed image when the "Bed on thumbnail" option is unchecked.

supermerill, 2021-06-08:
> For what it's worth, I just upgraded to 2.3.56.3 and I'm still seeing the bed image when the "Bed on thumbnail" option is unchecked.

I just checked and there is no more bed on my 3mf, small thumbnail and big thumbnail.

closed 2021-06-13 by supermerill