supermerill/SuperSlicer #1246: [BUG] Flow rate bug

qwewer0, 2021-06-04:
**Describe the bug**
Different flow applied to some parts of the print, where it should be the same as the surrounding area.

**Project File**
[bridge_angle_test_five_arms.zip](https://github.com/supermerill/SuperSlicer/files/6600670/bridge_angle_test_five_arms.zip)

**Expected behavior**
Flow rate should be consistent to the settings.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 2.3.56.2

supermerill, 2021-06-05:
I checked the gcode and nothing wrong here.
I'm more leaning toward a quirk in the code viewer when using relative extrusion, but I don't know why.

supermerill, 2021-06-05:
ok, it was a lack of precision in some important variables in the gcode parser (flat instead of double)

qwewer0, 2021-06-05:
> ok, it was a lack of precision in some important variables in the gcode viewer (float instead of double)

Thank you for the quick fix, now I cannot see that artifact, but I found some other flow/gcode view related artifacts, which for I will open a new issue to not confuse thing.

closed 2021-06-05 by qwewer0