supermerill/SuperSlicer #1247: 2.3.56.2 changes TAB`s with number keys

Jack77777777, 2021-06-04:
1, 2, 3, 4, 6, is changing the Tab´s.
It´s not possibl eto change the view´s anymore with the numbers and views are greyed out too.
It is also not possible to use this numbers in any setting.

CageFox, 2021-06-04:
No, it works like it adding Ctrl to any 1, 2, 3, 4, 6 in any field

I press 1 in text field. I receive Ctrl+1 effect: first tab focused

supermerill, 2021-06-04:
if I press 1 in start-gcode, it write "1", tested with main row and keypad on windows.

CageFox, 2021-06-04:
If I press 1 in start gcode it focus first tab "3D View" and no "1" inserted
If I press 2 in the same place it focus second tab "Sliced Preview"

Ctrl+1 and Ctrl+2 do the same things

supermerill, 2021-06-04:
@CageFox what is your susi version, your os?, your current susi gui configuration?

CageFox, 2021-06-04:
Yeah, I can enter digits from numpad ;) I just didn't tried this. 
But 1-4 and 6 from main part of keyboard works like Ctrl pressed. 5 works ok. 6 do nothing: no tab switch, no 6 inserted

supermerill, 2021-06-04:
@Jack77777777
keypad should works
you can revert to the old layout in preference -> gui, and you have access to the view items in this one.

for the "can't write number", I'll try on some other computers, but so far I have no problem typing them, so fixing this bug may be problematic.

CageFox, 2021-06-04:
> 
> 
> @Jack77777777
> you can revert to the old layout in preference -> gui, and you have access to the view items in this one.

Change of ui to old one changes nothing: the tabs changed to old, but pressing 1 still works like Ctrl+1

Even change ui to "Access via settings" with no tabs visible do not help

RMSMarc, 2021-06-04:
Same here, Windows 10 Pro. I don't have a numpad as a workaround.

CageFox, 2021-06-04:
> 
> 
> Same here, Windows 10 Pro. I don't have a numpad as a workaround.

The copy - paste also works. But the bug makes this version almost unusable.

randellhodges, 2021-06-05:
Windows 10 Pro 20H2, same problem as everyone else.

simpat1zq, 2021-06-05:
Just here to confirm. It does the same thing for me. Press the 1 through 6 keys on the main part of the keyboard changes to the corresponding tab. Using the numpad works as expected. 

If I change it to the old PS layout, it does the same thing. The other layouts do similar stuff as well. 

One thing I noticed is that number 1,2,3,4, and 6 change to the corresponding tab. However, the number 5 doesn't do anything. It even allows me to type the number 5 into settings dialogs. 

I'm using the US keyboard layout. Not sure if that is the culprit here.

randellhodges, 2021-06-05:
Looks like it is the cause. It is running even if the control key is not pressed, at least in windows. I noticed that the Ctrl+1 or Ctrl+F only works if you clicked inside an input field. I would sort of expect these shortcuts to work even if you don't actually have a cursor in an input field.

https://github.com/supermerill/SuperSlicer/blob/ce024979effc077aaf0c43405efe367a52472c18/src/slic3r/GUI/Field.cpp#L119-L129

randellhodges, 2021-06-05:
> Same here, Windows 10 Pro. I don't have a numpad as a workaround.

If you don't have a numpad, you can bring up the on screen keyboard, Win+Ctrl+O, then chose options, turn on numeric pad, and then click the numbers on that keypad and that'll work around it. A little easier than copy/paste

marcgiovannoni, 2021-06-05:
Numpad workaround is not working for me on the infill input on the 3D view tab. I can only select values from the dropdown.

Jack77777777, 2021-06-05:
> Change of ui to old one changes nothing: the tabs changed to old, but pressing 1 still works like Ctrl+1
> 
> 
> 
> Even change ui to "Access via settings" with no tabs visible do not help

Same for me

qwewer0, 2021-06-05:
I also see this on the [latest release](https://github.com/supermerill/SuperSlicer/releases/tag/2.3.56.2).
But on the latest Nigthly windows x64 [#337](https://github.com/supermerill/SuperSlicer/actions/runs/907997490) and [#339](https://github.com/supermerill/SuperSlicer/actions/runs/908201992) the numbers only changing the view.

supermerill, 2021-06-05:
@qwewer0 so the latest nigthly fix the issue?

can someone else test https://github.com/supermerill/SuperSlicer/suites/2916204183/artifacts/65458861 ? thx

Sirage48, 2021-06-05:
From macOS, only CMD 5 works now (fixed #1174)
[image]

Sirage48, 2021-06-05:
From 210604 notice the different format but only CMD 5 works. The others whilst here do not work
[image]

supermerill, 2021-06-05:
@Sirage48
please create a new issue next time, this one is for number keys not working.

can you check every menu entry for shortcut? particularly the edit & file one?

neophyl, 2021-06-05:
Merill I tried the nightly you linked to on my Windows 10 machine (Set to UK English keyboard).and it does seem to have fixed the number keys.
Using the top keyboard numbers they select the view angle in the plater, slice preview and gcode preview tabs.  Same with the number keys on the number pad.

The cntrl+number also changes the tabs correctly as far as I can see.

I was going to report that the release version also has a related issue, when trying to type notes into the filament profile every time I tried to type a 'j' it opened up the print queue which was most annoying.  However I also tried this in the nightly and that behavior has also been fixed.  

It does have a slight teensy tiny issue though, you can not type in the letter e in anymore.  Shift+e is allowed and will give you E or e depending on caps lock but e on its own doesn't work in any notes field (Printer, filament or print profiles).  I'll open up a new issue if you want.

supermerill, 2021-06-05:
> It does have a slight teensy tiny issue though, you can not type in the letter e in anymore. Shift+e is allowed and will give you E or e depending on caps lock but e on its own doesn't work in any notes field (Printer, filament or print profiles). I'll open up a new issue if you want.

will be fixed

CageFox, 2021-06-05:
> 
> 
> @qwewer0 so the latest nigthly fix the issue?
> 
> can someone else test https://github.com/supermerill/SuperSlicer/suites/2916204183/artifacts/65458861 ? thx

With this version digit keys works good

closed 2021-06-13 by supermerill