supermerill/SuperSlicer #1248: Can't open Superslicer SuperSlicer_2.3.56.2_macos_210604 with /datardir

Metti93, 2021-06-04:
**Describe the bug**
With the new version of SuperSlicer, I can no longer open the software with the command "/Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer --datadir /Volumes/home/3D-Print/SuperSlicer/config". It aborts every time and gives me an error

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Desktop'
2. Click on my custom Script "/Applications/SuperSlicer.app/Contents/MacOS/SuperSlicer --datadir /Volumes/home/3D-Print/SuperSlicer/config"
3. See error

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS Big Sur 11.4
 - SuperSlicer_2.3.56.2
**Additional context**
Add any other context about the problem here.

supermerill, 2021-06-04:
what was your previous working version?

Metti93, 2021-06-04:
SuperSlicer_2.2.56.0

supermerill, 2021-06-04:
I think the problem may come from a setting from the targeted setting directory.
Is it an empty one?
if not, can you zip it to me?

Metti93, 2021-06-04:
[config.zip](https://github.com/supermerill/SuperSlicer/files/6601653/config.zip)

Metti93, 2021-06-04:
I tried to change something else in the config. At first I thought it had something to do with the STL and the jpg for the PReview. 

which is why I have under 

bed_custom_model = 192.168.0.10\\Volumes\\home\\3D-Druck\\SuperSlicer\\config\\V0_120mm_Bed.stl
bed_custom_texture = 192.168.0.10\\Volumes\home\3D-printing\SuperSlicer\\config\voron_textured.png

once adjusted the address. Please ignore this

Translated with www.DeepL.com/Translator (free version)

supermerill, 2021-06-04:
I can replicate it, and I think it's already fixed in my dev build

Metti93, 2021-06-04:
it must have something to do with the --datadir. I can start the application this way if I don't apply the script. But when I want to retrieve the config from my NAS, the problem arises....

supermerill, 2021-06-05:
the problem is in the notification that say "hey there is a new profile to update"

closed 2021-06-13 by supermerill