supermerill/SuperSlicer #1249: Some Calibration options are crashing during slicing on Windows 10 Pro 64bit SuperSlicer 2.3.56.2

perfessor101, 2021-06-05:
### Version
_Version of SuperSlicer used goes here (help->about)_
Version 2.3.56.2

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 Professional 64 bit latest updates as of 2021 June 04

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_
  * _Screenshots if useful_
Most Calibration objects I get an 
[image]
I can slice downloaded STL's, but if under Calibration I try Bed Leveling, Filament Flow, Bridge Flow, Over Bridge Calibration, Top Flow Calibration the SuperSlicer crashes during slicing. 
These are the errors after clicking generate on Bed Leveling
A tool path outside the print area was detected and
An Object outside the print area was detected.
Resolve the current problem to continue slicing.
If I try Voron Cube it gets placed outside of build area.
_Is this a new feature request?_

Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/
not a feature request

#### Project File (.3MF) where problem occurs
can't upload project files as program crashes before I have a chance

_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

supermerill, 2021-06-05:
can I have your project file? it's to get your current settings, as I can't reproduce this with an empty printer

perfessor101, 2021-06-09:
it's crashing really fast ... not sure if I can save a project file before as it automatically slices for the calibration presets

I cleared my programdata/SuperSlicer directories and tried again with the previous and the latest SuperSlicer and there are no problems.

I should have a backup of the older settings, 
it may have been an amalgamation of too many settings old and new / changed

closed 2021-06-09 by perfessor101

supermerill, 2021-06-09:
Can you send me a zip of your old setting directory?