supermerill/SuperSlicer #1250: differences with Prusaslicer

wnc2, 2021-06-05:
**Describe the bug**
the same stl is sliced very different, with a lot of issues
**To Reproduce**

Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**>> Project File <<**

[Archive.zip](https://github.com/supermerill/SuperSlicer/files/6602314/Archive.zip)

**Expected behavior**
i should see a very similar result

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
MACOS
2.3.56

**Additional context**
i have this issue since a lot of releases

supermerill, 2021-06-05:
it's because the "only one perimeter on top" is detecting area too thin and so deactivate itself.
Deactivate the option.

wnc2, 2021-06-05:
[image]

this is the result, not identical but much better.

Thanks

supermerill, 2021-06-05:
if you want to get back overlapping perimeter, check the option. It will overextrude the letters a bit... nothing to worry in this case.

wnc2, 2021-06-05:
[image]

overlapping is checked by default, checking also "also for all perimeter" result is identical.

many thanks

closed 2021-06-14 by supermerill