supermerill/SuperSlicer #1255: [Windows] SS not using GPU

simpat1zq, 2021-06-05:
So I noticed that the Plater, Gcode and Sliced Previews were both running laggy. And it's worse with larger prints. And it's better if I have it show few layers using the vertical slider. I looked in the Windows task manager, and it's not using the GPU. I'm on the latest recently released version. 

The previous release uses a GPU and is perfectly smooth on all 3d views. 

Is there some setting I need to adjust to have it use the GPU? The GCode preview becomes close to unusable with I have a lot of things on the bed or I have larger things.

neophyl, 2021-06-05:
Sounds like a lot of issues reported over on the Prusa Slicer github for the 2.3 release.

supermerill, 2021-06-05:
version?
do you have a opengl32.dll in the root folder, ready to be deleted?

simpat1zq, 2021-06-05:
> 
> 
> version?
> do you have a opengl32.dll in the root folder, ready to be deleted?

Thanks. That was it. I deleted it and it's all good now.

closed 2021-06-05 by simpat1zq