supermerill/SuperSlicer #1256: Fill angle increment values don't work correctly anymore

neophyl, 2021-06-05:
Version: 2.3.56.2
Build: SuperSlicer_2.3.56.2_2021-06-05

**Describe the bug**
One of the things useful for the infill angle increment setting is to make sure all the infill goes in the same direction which is vital for getting clear prints.  Well as clear as they can be.  In Prusa Slicer you have to use 'aligned rectilinear' for top/bottom and infill at 100% to do the same job.

As Super Slicer doesn't have the aligned rectilinear infill any more the fill angle increment is the only way to do the same job with rectilinear or monotonic.  I was slicing for some clear parts earlier using the nightly as I had it open.  However to get the infill to be all the same direction you have to set the angle to 90 degrees.  Which is a strange value as either 180 or 360 would seem to be what you would enter if you wanted it all the same way.  So I did a bit of testing.

I tried various increment values to make it work and entering in something like 5 or 10 degrees will show you that it is not functioning as the initial angle is changed but then it alternates at 90 degrees rather than changing by the set angle.

I suspect its adding the increment angle to a 90 degree change at each layer.  So setting 90 is giving 180 and setting 10 degrees is giving 100.  

[Infill Increment.zip](https://github.com/supermerill/SuperSlicer/files/6602597/Infill.Increment.zip)

**To Reproduce**
open the attached project and slice.  then check the preview.

**Desktop (please complete the following information):**
Windows 10 64 bit

supermerill, 2021-06-05:
>doesn't have the aligned rectilinear infill any more

what version have it?

> I suspect its adding the increment angle to a 90 degree change at each layer. So setting 90 is giving 180 and setting 10 degrees is giving 100.

yes, and setting it to 100 should creates a change of 10°  each layer

neophyl, 2021-06-05:
I’d have to check to be absolutely sure but from what I remember if you set 5 degrees it just rotated the pattern by 5 degreees and not 95. The initial angle still worked the same and then it just incremented it. 

Sorry, I thought susi had that pattern as well as PS, wonder when that was added to PS.

supermerill, 2021-06-05:
> Sorry, I thought susi had that pattern as well as PS, wonder when that was added to PS.

maybe i didn't merge it because it's just rectilinear with a 90° increment angle.

rs-development, 2022-12-05:
I think you are missing that with a 90° increment angle also the solid infill is aligned.

Aligned rectlinear prints very clean as an sparse infill, so i would like to use it for that. But the solid infill still needs to be rotated with each layer for a better closing of the sparse infill.

supermerill, 2023-12-28:
now there is even angle template. Should be easier.

closed 2023-12-28 by supermerill