supermerill/SuperSlicer #1259: 2.3.56.2 does not compile

Lukas1818, 2021-06-05:
### Version
2.3.56.2

### Operating system type + version
Arch Linux and Manjaro

### Behavior
While updating the superlicer aur package, I notify that the current release does not compile:
```
In file included from /home/lukas/git/aur/superslicer-prerelease/src/SuperSlicer-2.3.56.2/src/slic3r/GUI/OptionsGroup.hpp:14,
                 from /home/lukas/git/aur/superslicer-prerelease/src/SuperSlicer-2.3.56.2/src/slic3r/GUI/Preferences.cpp:2:
/home/lukas/git/aur/superslicer-prerelease/src/SuperSlicer-2.3.56.2/src/slic3r/GUI/Preferences.cpp: In member function 'void Slic3r::GUI::PreferencesDialog::create_settings_mode_widget()':
/home/lukas/git/aur/superslicer-prerelease/src/SuperSlicer-2.3.56.2/src/slic3r/GUI/I18N.hpp:3:47: error: conversion from 'wxString' to non-scalar type 'std::wstring' {aka 'std::__cxx11::basic_string<wchar_t>'} requested
    3 | #define _L(s)     Slic3r::GUI::I18N::translate((s))
      |                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
```

My ci hit the same problem at the -git package, which does build the same commit at the moment:
https://gitlab.com/Lukas1818/aur-package-check/-/jobs/1320724939#L2120

JohanAR, 2021-06-05:
I think it's commit aaefb499674ce602bc9e8323c9d2949411c0205f

Possibly because it changes an #ifdef __APPLE__ into an #ifndef WIN32, which might accidentally changed the behaviour for Linux, but I haven't looked into it

supermerill, 2021-06-05:
i know, i've fixed it.

Lukas1818, 2021-06-05:
> i know, i've fixed it.

Thanks for the fast fix, but where have you fix it? I do not find a new commit, which fix this issue.

supermerill, 2021-06-05:
c0485ec817aafcbc427a41db1ec47695f1fc53ab

Lukas1818, 2021-06-05:
> [c0485ec](https://github.com/supermerill/SuperSlicer/commit/c0485ec817aafcbc427a41db1ec47695f1fc53ab)

superslicer-git does build the current master and still hit this problem. So it seems that the commit does not fix this issue.

Lukas1818, 2021-06-05:
> > [c0485ec](https://github.com/supermerill/SuperSlicer/commit/c0485ec817aafcbc427a41db1ec47695f1fc53ab)
> 
> superslicer-git does build the current master and still hit this problem. So it seems that the commit does not fix this issue.

It seems I did accidentally build the false superslicer package. Sorry for this.

But I hit another issue

closed 2021-06-05 by Lukas1818