supermerill/SuperSlicer #1260: [BUG] Wrong minimum limit value in legend

qwewer0, 2021-06-05:
**Describe the bug**
SuperSlicer shows wrong (lower) minimum limit value for flow in legend, than what is used in the print, which is different in the gcode viewer.

**Project File**
[bridge_angle_test_five_arms.zip](https://github.com/supermerill/SuperSlicer/files/6602699/bridge_angle_test_five_arms.zip)

**Expected behavior**
The minimum and maximum values in legend should come from the sliced project.

**Screenshots**
Gcode viewer:
[image]

Inside SuperSlicer:
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 2.3.56.2 (Nigthly windows x64 [#5](https://github.com/supermerill/SuperSlicer/actions/runs/909270246))

supermerill, 2021-06-07:
I have min at 2.4 in 2.3.56.2
I have mina t 3.1 in dev build

So maybe fixed...

qwewer0, 2021-06-07:
> I have min at 2.4 in 2.3.56.2
> I have mina t 3.1 in dev build
> 
> So maybe fixed...

Yes, it seems to be fixed, thanks.

closed 2021-06-07 by qwewer0