supermerill/SuperSlicer #1261: Languages display error

sziga, 2021-06-05:
There are problems with the display of languages.
Even though it is translated and included in the language files (.po, .mo), many of the terms appear in English.
E.g. "Andvanced", "Remember to check for updates at" and many more.
Version:SuperSlicer_2.3.56.2_win64_210604 but the problem persists with the previous version.
Example images in Italian, Czech and Japanese.
[image]
[image]
[image]

I put the Italian language file into PrusaSlicer-2.3.1 and it displays everything in Italian without any errors.
[image]

supermerill, 2021-06-05:
That's because the translations weren't updated. I don't speak japanese.

sziga, 2021-06-05:
Are typed in Italian .po file and do not appear the same! the file is moved to PrusaSlicer (see picture 4) to display the expressions.
Italian .po file
Line 736
msgid "Advanced"
msgstr "Avanzato"

Line 7876
msgid "Remember to check for updates at "
msgstr "Ricorda di controllare gli aggiornamenti dell'applicazione"

supermerill, 2021-06-05:
please use the thread #1128 to talk about improving the italian translation.
I work a bit to ease a bit the burden to update a translation (I'm working on the french one when I'm disconnected from the internet). But it's not my absolute priority.

closed 2021-06-14 by supermerill