supermerill/SuperSlicer #1263: macOS Shortcuts

Sirage48, 2021-06-05:
Replacing my comments on #1247.  As requested:
1. File shortcuts all seem to work and layout the same 56.1 56.2 56.3 (version number still 56.2)

2. Edit shortcuts also still work BUT they have disappeared in 56.3
56.2
[image]

56.3
[image]

3. The TAB shortcuts not working except Filament.

4. Under View "Collapse sidebar"  Shift-TAB is unstable and now flickers collapsed - not collapsed and will finish on one or the other depending how many milliseconds you hold the keys.

supermerill, 2021-06-06:
Yes, i wondered if I should keep "CTRL" or change them with "CMD"
Can i have a list of working shortcut and a list of broken one?

supermerill, 2021-06-09:
@liffbag
can someone report on what's showing in the menus with  version https://github.com/supermerill/SuperSlicer/actions/runs/922009616 ?
I should have
```
### File
new project :		Ctrl+N
Open project:		Ctrl+O
save project:		Ctrl+S
save project as...:	Ctrl+Alt+S
Import->stl...:		Ctrl+I
Import->config:		Ctrl+L
Import->config from project:	Ctrl+Alt+L
export->ExportGcode:Ctrl+G
export->Send Gcode .. :	Ctrl+Shift+G
export->gcode to sd:Ctrl+U
export-> config:	Ctrl+E
### Edit
Select all:			Ctrl+A
Deselect all:		Esc
Delete selected:	Del
Delete all:			Ctrl+Del
undo:				Ctrl+Z
Redo:				Ctrl+Y
Copy:				Ctrl+C
paste:				Ctrl+V
reload from disk:	F5
Search:				Ctrl+F
###Windows
tabs: Ctrl+1 to Ctrl+6
Print host upqueue:	Ctrl+J
Open new instance:	Ctrl+shift+I
###View
change viewpoint: 0 to 6
show labels:		Ctrl+E
colapse sidebar:	Shift+Tab
###Configuration
preference			Ctrl+P
```

With a screenshot of the edit menu plz.

macdarren, 2021-06-09:
[image]

macdarren, 2021-06-09:
[image]

[image]

[image]

[image]

supermerill, 2021-06-09:
can someone test if they are working in this build?
https://github.com/supermerill/SuperSlicer/suites/2955019072/artifacts/66525953
C/C++ Nigthly macos #323:

Sirage48, 2021-06-10:
Preliminary checks indicate nearly every shortcut is now working - thanks.
macOS Big Sur 11.3.1

The only one not working so far is E (View / Show Labels) which is flakey. Either very slow, many seconds, or not responsive. Not a priority to fix I guess.

⌘ 1 - 6 all good
Shift - Tab working which might fix #1266 ??? which was on 11.4 that I think is a development build of Big Sur.

I noticed preferences are ⌘,  vs ⌘P in Prusaslicer which is shown when pressing   "?"
Just an observation - not a request to change.

I will continue.

Sirage48, 2021-06-10:
I have been through all the 6 Shortcut help screens on macOS and they all seem to work except some of the Gizmos, which I don't usually use so I don't care too much about these.  In the interest of completion or perfection (??):

- ⌘ for one direction did not work

- ⌥ in conjunction with S  scale or R rotate seems redundant as the default operation uses the centre anyway
       A mystery,  someone else might help (or educate me).

In summary so far all the obvious shortcuts now OK on my Mac with nightly 323  = 2.3.56.4

supermerill, 2021-06-10:
preferences shortcut info changed to `cmd+,`

> The only one not working so far is E (View / Show Labels) which is flakey. Either very slow, many seconds, or not responsive. Not a priority to fix I guess.

well, that's another problem. Is it the same with ps 2.3.1?

Sirage48, 2021-06-10:
E does work fine in ps 2.3.1

Sirage48, 2021-06-10:
Just experimented with the nightly 323 and the latest rc.
E might work initially after a second or so, and then if repeated it may work after 5 or 10 seconds.
But if I press E followed by clicking the mouse anywhere it works instantly both for show or hide labels.
Just clarifying - not a priority.

Sirage48, 2021-06-11:
E fixed in #325
Excellent.

closed 2021-06-26 by supermerill