supermerill/SuperSlicer #1266: macOS Sliced & Gcode Tabs (& Details Pane)  Blank after a slice

jallen810, 2021-06-06:
Basically the main window goes blank after I click the slice button. This has been an issue for me for the last 6 or so versions of S.S. I can remedy the issue by simply clicking to the "Print"/"Filament"/ or "Printer" settings tabs and then back to the "Sliced" or "Gcode" tabs
As for the side bar (or detail panel), by simply hiding/un-hiding, it will then generate the data.

See attached video screen record, link here: 
https://www.icloud.com/iclouddrive/0-AMcP48d9EXTexqsvcxOYYrA#SuperSlicer_macOS_Blank_Screen_Bugs

I am running a MacBook Pro 16" 2020, macOS Big Sur 11.4, i7 2.6GHz 6-core, 16gb DDR4, AMD Radeon Pro 5300M 4 GB
Intel UHD Graphics 630 1536 MB

SuperSlicer Version 2.3.56.1
[image]

[image]

supermerill, 2021-06-06:
weird.
Does it work if you're using the "old" GUI layout?

supermerill, 2021-06-09:
Can't really resolve this issue without hours and hours of debug to understand how the gui library  work & react on a real amc.

Sirage48, 2021-06-10:
I can't reproduce this behaviour. Maybe requires a 3mf to test.

supermerill, 2021-06-10:
> I can't reproduce this behaviour. Maybe requires a 3mf to test.

I changed somethign (and fix a line) in the last nightly, so it may be fixed.

jallen810, 2021-06-16:
> weird.
> Does it work if you're using the "old" GUI layout?

No, unfortunately not. I've tried every layout and nothing seems to help.

jallen810, 2021-06-16:
> I can't reproduce this behaviour. Maybe requires a 3mf to test.

I can tell you in my experience that it does not matter what the part is, ever single time it does it, but here is a .3mf for you... Perhaps it's got something to do with my GPU???? 
[Testpart_forSSBug.3mf.zip](https://github.com/supermerill/SuperSlicer/files/6660095/Testpart_forSSBug.3mf.zip)

jallen810, 2021-06-16:
> > I can't reproduce this behaviour. Maybe requires a 3mf to test.
> 
> I changed something (and fix a line) in the last nightly, so it may be fixed.

I am currently running version 2.3.56.5 and it is still present unfortunately. I will say that this is not a "show stopper" but it's definitely very annoying. Thanks for looking into it

jallen810, 2021-06-20:
> > I can't reproduce this behaviour. Maybe requires a 3mf to test.
> 
> I changed something (and fix a line) in the last nightly, so it may be fixed.

I downloaded the nightly version, and the bug is present in the "Regular Layout with Tab Bar" 
HOWEVER, if you switch to the "Old PrusaSlicer Layout" option, the bug goes away and all functions normally. If your change was related to this, it works =)