supermerill/SuperSlicer #1268: bridge flow calibration crash when bridge_flow_ratio value is not a multiple of 5

kevenwyld, 2021-06-07:
**Describe the bug**
When `bridge_flow_ratio` is not set to a multiple of 5, for example 108%, running the bridge flow calibration fails with an error and SuperSlicer crashes/exits.

**To Reproduce**
Steps to reproduce the behavior:
1. Set `bridge_flow_ratio` under Print Settings -> Width & Flow -> Flow ratio -> Bridge to 108%.
2. Select the bridge flow calibration from the drop down (Calibration -> Bridge Flow Calibration)
3. Click "Test Flow Ratio" with any values
4. Crash.

**>> Project File <<**
No project file, this is for the built in calibration

**Expected behavior**
Not a crash

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Archlinux
 - SuperSlicer Version 2.3.56

**Additional context**
This seems to be because the bridge flow calibration looks for a file which matches the current value in `/usr/share/SuperSlicer/calibration/bridge_flow/` on Linux.  The specific amf files are required to print the percentages on the calibration objects.   If it were me I'd adjust this to be like the temperature tower where values below 180 and above 285 (and values in between if you entered them) are not printed but the test is still completed.

qwewer0, 2021-06-08:
Slicer is no longer crashed, so this issue can be closed.

But if a flow used that isn't a multiple of 5, then no number is generated.
@supermerill What do you think if in those cases a blank block would be used instead of nothing?

[image]

supermerill, 2021-06-08:
maybe, or I have to create single-digit amf and combine them to create the numbers. it's just a pain in the ass to rotate & translate them correctly.

qwewer0, 2021-06-08:
> maybe, or I have to create single-digit amf and combine them to create the numbers. it's just a pain in the ass to rotate & translate them correctly.

I didn’t want to suggest it because - as you too wrote it - I thought it would be a pain in the ass, but the single-digit amf would be the best solution, and nothing further would need to be made to the digits.

closed 2021-06-13 by supermerill