supermerill/SuperSlicer #1270: GUI issues related to spacing settings

JohanAR, 2021-06-07:
**Describe the bug**
See comments on screenshots

**Screenshots**
After starting SuperSlicer 2.3.56.3 with my existing profiles:
[image]
1. "spacing" is written with black text
2. "support material" and "skirt" are written with green text, while the others are white. It looks quite arbitrary, but is there some special meaning of green fields?

After resetting spacing by clicking the blue padlock:
[image]
1. "width" is now black
2. "spacing" turns orange
3. skirt resets to "120%" despite the unit type is only listed as "mm"

**Desktop (please complete the following information):**
 - OS: Linux
 - Version: 2.3.56.3 and also on current master

supermerill, 2021-06-07:
For the signification of green and the other colors & icon, clic on the `?` icon

As the system value may ahve been changed with the new version, lock & green text may have changed at the same time.

JohanAR, 2021-06-07:
Sorry, I guess I was a bit confused by never having seen black before, and I forgot that they turn orange sometimes..

I guess the only remaining error is that skirt says that it can only be "mm" while it ought to be "mm or %" then?

supermerill, 2021-06-07:
yes

closed 2021-06-13 by supermerill