supermerill/SuperSlicer #1272: Supports being printed in mid air

ashleyuk90, 2021-06-07:
Have just tried out SuperSlicer from using Prusa Slicer previously, using the same support settings (have tried all 3 support Patterns) I have noticed SuperSlicer in quite a few places is not building the correct support structure and instead is trying to print the support material interface in mid-air.

PrusaSlicer
[image]

SuperSlicer
[image]

Support Settings
[image]

qwewer0, 2021-06-07:
https://github.com/supermerill/SuperSlicer/issues/1199 Related?

ashleyuk90, 2021-06-07:
> 
> 
> #1199 Related/duplicate?

Maybe, but my understanding from that bug the issue also showed in PS? I am running the same version of PS that was recently merged into SS. So it seems more something custom/change in SS that's breaking it compared to the still issues that PS are working on with their supports in general.

supermerill, 2021-06-07:
please use the thread 1199 .

closed 2021-06-07 by supermerill