supermerill/SuperSlicer #1273: Using the same Top and Bottom Z-offset, support are more difficult to remove in 2.3.56.3 than in 2.2.53

laptop-laptop, 2021-06-07:
**Describe the bug**

Using the same Top and Bottom Z-offset values, why support became more difficult to remove in 2.3.56.3 than in 2.2.53?

Also, with Support Everywhere, under 2.3.56.3, large elephant foot-like stuff is created in the first layer. No such thing under 2.2.53

supermerill, 2021-06-07:
> Using the same Top and Bottom Z-offset values, why support became more difficult to remove in 2.3.56.3 than in 2.2.53?

not enough data to answer.

> Also, with Support Everywhere, under 2.3.56.3, large elephant foot-like stuff is created in the first layer. No such thing under 2.2.53

Do you have a project file to show me the problem? (and maybe pictures) I don't understand.

Znubbis, 2021-07-22:
I have the same symptoms that what i think is called material interface layer is fused with the layer under it making it impossible to separate. I will try set the bottom z offset to 0.3mm to check if that helps.
This happens when using support on already printed areas, when using support directly on the bed i assume it will work fine.