supermerill/SuperSlicer #1274: Some bugs in version 2.3.56.3 for macOs

erathos77, 2021-06-07:
New install of version 2.3.56.3 for macOs (10.13):
-menu Help / system info: remains rated version 2.3.56.2 instead of 2.3.56.3
- in french language version: when generating gcode the lower strip of the window shows in French « découpe annulée » (= slice canceled) although the gcode file obtained  is OK.
in English language version: that lower strip shows « slicing complete » after generating gcode successfully but…
- superslicer freezes and must be forced to quit after exporting gcode (regardless of language version). The gcode file is OK however.
-menu window/gcode preview tab does’nt show any gcode preview

supermerill, 2021-06-08:
> remains rated version 2.3.56.2 instead of 2.3.56.3

reported at #1271

> superslicer freezes and must be forced to quit after exporting gcode (regardless of language version).

seems identical to #1278
can't really debug as my version don't hang... it's hanging before showing the gcode preview? what gui layout are you using?

erathos77, 2021-06-08:
It's hanging only AFTER the export of gcode to the desktop when slicing is complete. I tried to re-download, re-install, or re-start  without success. Might be a system problem (macOs 10.13 or 10.14) ? (although Slic3r 1.3 works fine)

After importing any stl object, the gcode preview tab triggers slicing and shows the sliced object on the new layout, but no gcode preview. I can only get the tool position by moving the lower arrow (am I wrong by waiting the display of true gcode text ?) .
Not sure to understand what do you mean by gui layout. In preferences/GUI I have chosen regular layout with the tab bar.

supermerill, 2021-06-08:
> Not sure to understand what do you mean by gui layout. In preferences/GUI I have chosen regular layout with the tab bar.

That's that. The tooltip should warn you about possible problem with some os.
Do you have hanging if you use the "old prusaslicer" layout ?

liftbag, 2021-06-08:
> > Not sure to understand what do you mean by gui layout. In preferences/GUI I have chosen regular layout with the tab bar.
> 
> That's that. The tooltip should warn you about possible problem with some os.
> Do you have hanging if you use the "old prusaslicer" layout ?

I can confirm the issue on macOS, it freezes after that the g-code is exported. Old prusaslicer GUI. 
The issue is also present in the nightly after the last release.
Launching the program from the terminal in loglevel 5 gives no information about any errors.
I tried different stl and different print and filament settings, always with the same result.
I doubt it will be useful, but I am attaching the a 3mf project that is causing the problem on macOS.

[Cerreto_Table_Tilter.3mf.zip](https://github.com/supermerill/SuperSlicer/files/6619291/Cerreto_Table_Tilter.3mf.zip)

erathos77, 2021-06-08:
Yes, according to liftbag's comment, same hanging issue using old prusaslicer layout.

erathos77, 2021-06-09:
for me it freezes on the first round so do i have to force to quit before moving on to the next stl.

supermerill, 2021-06-09:
please use  #1278 for hanging issue

closed 2021-06-09 by supermerill