supermerill/SuperSlicer #1275: [FR] Gap Fill Using Fixed Volumetric Rate

CCS86, 2021-06-08:
Printers (especially bowden) struggle with rapid changes to commanded volumetric flow rate. Gap filling moves have a tendency to create large volumetric rate changes, across small distances. Also, my gut says that it is harder to fill a 0.8mm wide gap, at some given flow rate, than it is to lay down a 0.4mm line, at the same flow rate (0.4mm nozzle). This would further exacerbate the issue of under-extruding in high flow + wide gap fill areas.

The tool tip recommends setting a very slow gap filling speed, to avoid this very issue. But, it creates a new one, which is over-extruding very thin gap fill areas. 

I think that allowing the user to define a volumetric rate to fill gaps would solve both. 

It would make sense to put a speed maximum against this calculation as well.

[image]

qwewer0, 2021-06-08:
> But, it creates a new one, which is over-extruding very thin gap fill areas.

I also experiencing it, and always felt weird that the gap fills were overextruding on my well tuned (direct drive) system.

> I think that allowing the user to define a volumetric rate to fill gaps would solve both.

Not sure if I understand it correctly, do you mean, to have an option to control the gap fill flow rate as a percentage?

CCS86, 2021-06-08:
Not as a percentage, but as a fixed volumetric flow rate.

[Layer height] x [line width] x [speed] = volumetric flow rate

So, if most of your print is 0.2 x 0.4 x 60 = 4.8 mm^3/s, you might want to define gap filling to hold that rate.

Each segment of gap fill would work from that defined flow rate, to calculate the segment speed. A 0.2mm wide gap fill would print at 120 mm/s, and a 0.8mm wide segment would print at 30 mm/s

CCS86, 2021-06-08:
Correct. At least as an option.

CCS86, 2022-04-20:
Hey @supermerill , just wanted to ping on this request and see where it stands. I think it would really help the fact that we still have gap fills that range from very wide to very thin.

supermerill, 2022-04-28:
There is currently max volumetric (from filament) and speed.
If you put a high speed, it should follow the filament max volumetric, but on the small extrusions where it should be capped by the speed.

What I can add is a max volumetric acceleration (mm3/s2) to reduce the speed if the volumetric speed is changing a lot. But this may need a substantial change in the gcode code.

CCS86, 2022-04-28:
> There is currently max volumetric (from filament) and speed. If you put a high speed, it should follow the filament max volumetric, but on the thin extrusions where it should be capped by the speed.
> 
> What I can add is a max volumetric acceleration (mm3/s2) to reduce the speed if the volumetric speed is changing a lot. But this may need a substantial change in the gcode code.

I think the biggest issue with that approach is if you want to run a significantly higher volumetric flow rate on sparse/solid infill to save time. Inherently, you will have a lower flow rate in your perimeters when looking for good quality, and matching the perimeter flow rate would likely be better for gap filling.

Would it be easier to just allow gap filling to have a 3rd definition type in the same input field (%, mm/s, mm^3/s)? or even just a checkbox to "match perimeter flow rate". When checked, it would calculate the perimeter flow rate, and use that with the width of each gap fill segment to calculate segment speed (capped by the max defined print speed).

supermerill, 2022-04-28:
the "match perimeter flow rate" is a good idea.

closed 2022-07-14 by CCS86

CCS86, 2022-07-23:
Oh man, this new setting is next level!!!

Thanks again Merill!

[image]