supermerill/SuperSlicer #1277: [BUG] No perimeters on bridge areas

qwewer0, 2021-06-08:
**Describe the bug**
There seems to be something wrong with `No perimeters on bridge areas` setting.

**Project File**
[bridge_angle_test_five_arms.zip](https://github.com/supermerill/SuperSlicer/files/6613128/bridge_angle_test_five_arms.zip)

**Expected behavior**
`No perimeters on bridge areas` settings should work as intended.

**Screenshots**
`No perimeters on bridge areas` setting:
- Disabled:
[image]
- Remove perimeters:
[image]
- Keep only bridges:
[image]
- Keep bridges and overhangs:
[image]
- Fill the voids with bridges:
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 2.3.56.3 (Nigthly windows x64 [#363](https://github.com/supermerill/SuperSlicer/actions/runs/915714312))

supermerill, 2021-06-08:
bridge detection for this feature seems to works for the whole object layer, and so can't use different angle for each "bridge island".
I already see some same strange behavior here #1157
Big work.

qwewer0, 2021-11-07:
Fixed on the latest [2.3.57.4](https://github.com/supermerill/SuperSlicer/releases/tag/2.3.57.4).

@supermerill Thanks for the fix, it looks perfect.

closed 2021-11-07 by qwewer0