supermerill/SuperSlicer #1278: SS is hanging if I use it to create more than 1 gcode for different stl's without quitting first

darrenph1, 2021-06-08:
**Describe the bug**
If I open SS and add an stl  and export it's fine. But if I then delete that stl, add another and change filament and/or print settings. It hangs after creating the new gcode.

**To Reproduce**
Steps to reproduce the behavior:
1. Add stl and generate gcode
2. delete stl and add different stl and change parameters (filament/ printer)
3. Save stl and it hangs after it generates it.`
4. See error

**>> Project File <<**
n/a

**Expected behavior**
It should not hang

**Screenshots**
n/a
**Desktop (please complete the following information):**
 - OS: Mac 10.14.6
 - Version 2.3.56

supermerill, 2021-06-08:
can you put here the project you use for the first print & the second one?
Does it hang is I just load the second project?
is it the 2.3.56.3?

darrenph1, 2021-06-08:
Not sure what changed but I can't replicate it this morning. I will test on my other machine and if nothing will close it out. Might of just been my mac.

darrenph1, 2021-06-08:
OK I had it happen again. I was using the attached project and exported a gcode. Then put then deleted the stl and put in my lize z stl and changed settings. It exported the included gcode and then froze.
[Freeze.zip](https://github.com/supermerill/SuperSlicer/files/6618969/Freeze.zip)

liftbag, 2021-06-09:
You are still one step ahead of me. On my mac (and also to other users), the latest release hangs on the first export of the g-code. When that happens, SuperSlicer process takes up 100% of the CPU.
I also wrote about it here #1274 

iMac (Retina 5K, 27-inch, 2017)
macOS Big Sur 11.2.2

supermerill, 2021-06-09:
> OK I had it happen again. I was using the attached project and exported a gcode. Then deleted the stl and put in my lize z stl and changed settings. It exported the included gcode and then froze.
> [Freeze.zip](https://github.com/supermerill/SuperSlicer/files/6618969/Freeze.zip)

Can't replicate on windows.
I'm going to output many mac builds to be able to get the commit where this behavior began to appear.

supermerill, 2021-06-09:
Can someone test 
https://github.com/supermerill/SuperSlicer/suites/2950257866/artifacts/66379546
https://github.com/supermerill/SuperSlicer/suites/2950276761/artifacts/66381139
https://github.com/supermerill/SuperSlicer/suites/2950294531/artifacts/66381053
and tell me the one(s) that won't freeze / hang if any

liftbag, 2021-06-09:
@supermerill 

66379546 don't hangs.

66381139 crashes on start up with the following log: 
[66381139_crash.zip](https://github.com/supermerill/SuperSlicer/files/6624054/66381139_crash.zip)

66381053 don't hangs.

liftbag, 2021-06-09:
Also the Nigthly macos #318 don't hangs.

supermerill, 2021-06-09:
> Also the Nigthly macos #318 don't hangs.

erf, why did i bother?

liftbag, 2021-06-09:
> erf, why did i bother?
We did not understand the nature of the issue, but all in all, it's okay!

liftbag, 2021-06-09:
The nightly #319 also don't hangs, and solves the issue #1271.

Aneurin, 2021-06-09:
Just to confirm in case anyone else comes here looking for this problem: the same thing happens on Linux, and it's similarly fixed in the latest nightlies (I used https://github.com/supermerill/SuperSlicer/actions/runs/921723200).

supermerill, 2021-06-09:
Just to be sure, as I added some fixes, is this version working on macos? And with the "tab layout"?
can someone test if they are working in this build?
https://github.com/supermerill/SuperSlicer/suites/2955019072/artifacts/66525953
C/C++ Nigthly macos #323:

macdarren, 2021-06-10:
Seems to work and slice for me.... Intel MacBook Pro running BigSur.
GUI is still weird with blank screens  etc in the 'regular layout' but using the Classic Prusa it is ok.
This has been true with most of the latest releases, so I use the Prusa layout by default.

closed 2021-06-13 by supermerill