supermerill/SuperSlicer #1279: got error a toolpath outside the print area

tjengbudi, 2021-06-08:
**Describe the bug**
[image]
[image]

latest release test with attached stl have error a toolpath outside the print area. i try slice with prusa slicer it is not got the same error. i already try to fix the stl and still the same. i also check to import the setting from prusa but still got the error

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
[1. X axis top.zip](https://github.com/supermerill/SuperSlicer/files/6613438/1.X.axis.top.zip)
2. Click on 'slice'
4. See error

supermerill, 2021-06-08:
[image]

can't reproduce, can you post here the project file (.3mf) instead of the stl?

tjengbudi, 2021-06-08:
tried again after reboot. download superslicer again. put the object like the first picture. it is working. do not know why. i trying slicing several time before. sorry, it might cause by my pc

closed 2021-06-08 by tjengbudi