supermerill/SuperSlicer #1284: Scrolling through menus changes setting values

ilikecandy, 2021-06-09:
**Is your feature request related to a problem? Please describe.**
When scrolling through settings it sometimes changes the values.

**Describe the solution you'd like**
Maybe like an option in settings to enable or disable it.

**Describe how it would work**
Toggle it to on and it would be exactly how it is now, disable it and scrolling will not affect the options

**Describe alternatives you've considered**

**Additional context**

supermerill, 2021-06-09:
> When scrolling through settings it sometimes changes the values.

if you click on a combobox (the widget with an arrow that display a list) and then scroll with the mouse?
If it's that, all software do that, I just test libreoffice and it's the same.
If it's not that, I need a bit more context.

ilikecandy, 2021-06-09:
[image]
 Just scrolling, no click

BlueDragonX, 2021-06-09:
What's actually happening here is hovering the mouse over a UI control selects that UI control. This isn't normal behavior in Windows. However, PrusaSlicer does the same thing. I've personally found it particularly annoying when copy/pasting X/Y/Z values in the plater.

supermerill, 2021-06-10:
> [image]
> Just scrolling, no click

like in libreoffice and other software...

supermerill, 2024-05-23:
looked at it, it's hardcoded in the gui library.
So it needs a patch. meh.