supermerill/SuperSlicer #1285: No Z-Hop for Gap Fill in Top Infill

BlueDragonX, 2021-06-10:
**Describe the bug**
No z-hop is present to or from the gap fill in solid infill even though a retraction is present prior to the travel moves. This causes nozzle scarring since the travel moves happen across already printed solid infill.

Here you can see the fill happening in a curved corner of solid infill:
[image]
You can see the travel move coming to the gap fill did not have a z-hop.

Here you can  see the retraction and travel move leaving the gap fill:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Enable retractions. I have configured firmware retractions in Klipper and enabled "Use Firmware Retractions" in SuperSlicer.
2. Enable z-hop. I have configured 0.4mm z-hops and selected "Only on Top".
3. Ensure retractions are enabled where possible. I have enable "Retract on Layer Change" and disabled "Only Retract when Crossing Perimeters".
4. Configure "Monotonic (filled)" for the infill.
5. Slice a part which requires gap fill on the top surface and has a travel move afterwards.

**>> Project File <<**
[Filament Card Z-Hop Test.zip](https://github.com/supermerill/SuperSlicer/files/6627514/Filament.Card.Z-Hop.Test.zip)

**Expected behavior**
I expect to see a z-hop on the travel before and after the gap fill.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 2.3.56.0

supermerill, 2021-06-10:
mmm... "only on top" means "only on extrusions marked as top solid infill" and it seems like the gapfill from the fill isn't taggued correctly in that regard.

BlueDragonX, 2021-06-14:
This is fixed for me using 2.3.56.5.

closed 2021-06-14 by BlueDragonX