supermerill/SuperSlicer #1286: Arrange spacing seems like it can't accept values below 4

simpat1zq, 2021-06-10:
I right click the arrange button, and move the slider. If it is above 4 and I move it, then click arrange, the objects will move. However, if I set the slider to 4 or below, arrange, and then move it and click arrange, nothing happens. I'm pretty sure that 4 is the minimum for some reason. It would be nice to allow it to go lower than that. I'm pretty sure that on some past versions I was able to set it to 1, and it worked as expected.

qwewer0, 2021-06-10:
For me the minimum arrangement distance is 6 mm, so it might correspond to a setting.
[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/6628161/Shape-Box.zip)

neophyl, 2021-06-10:
Isn’t there a spacing distance setting in the print settings ?  I suspect it’s acting as a minimum if it’s still in there. I’ll have a look when I get back to a pc.

supermerill, 2021-06-10:
I'll see if the gizmo can have his min changed.

simpat1zq, 2021-06-10:
> 
> 
> Isn’t there a spacing distance setting in the print settings ? I suspect it’s acting as a minimum if it’s still in there. I’ll have a look when I get back to a pc.

Ah. That was it. I changed it to 1mm. I'm guessing that this doesn't doesn't need any major change. You might just be able to put a little note under the slider to say that the minimum distance is defined there.

supermerill, 2021-09-06:
Now you can set it to 0, the min_distance from the print setting is now only used as the default value when starting the app.

closed 2021-09-06 by supermerill