supermerill/SuperSlicer #1288: Changing Layer Height Causing Skipped Perimeters

CCS86, 2021-06-10:
2.3.56.3 / Win10 x64

not sure what is going on here. Simply changing the base layer height from 0.15mm to 0.10mm is causing skipped perimeters in some areas, like the top of the print:

0.15mm layers:

[image]

0.10mm layers:

[image]

[spiral_cube_cube.3mf.txt](https://github.com/supermerill/SuperSlicer/files/6628024/spiral_cube_cube.3mf.txt)

qwewer0, 2021-06-10:
I wasn't able to reproduce the issue on the latest [Nigthly windows x64 #371](https://github.com/supermerill/SuperSlicer/actions/runs/922888913)
It only left maximum one perimeter empty, because gap fill was off.

0.1:
[image]

0.15:
[image]

supermerill, 2021-06-10:
can't replicate with the 2.3.56.3 neither...

CCS86, 2021-06-10:
Wow, that is bizarre.

I still had SS open, and loaded the project file I uploaded. Issue was present upon slicing. Perimeters were set to 20. I set them to 2, and got solid infill. set back to 20 and got the gap again.

Then I closed SS, relaunched it, opened the project... issue gone. Cannot get it to reproduce so far. It must have been some sequence of setting changes that led SS into a weird state.

qwewer0, 2021-06-10:
@CCS86 If you manage to reproduce it consistently, and share the steps you took, then I would like to test it.