supermerill/SuperSlicer #1289: incorrect labels?

jltx1, 2021-06-10:
### Version
2.3.56

### Operating system type + version
macOS 10.13

### Behavior
labels on feed rates under printer settings/machine limits say mm/min but act like mm/s.  Or they are not working to limit speed.

supermerill, 2021-06-10:
I don't know your firmware, so maybe yes, maybe no.
so maybe duplicate of #1184 btw

jltx1, 2021-06-10:
Oh, I didn't realize the labels change with firmware setting.  That explains how I got there (copying someone else's profile).  

So then my second statement still stands.  The machine limit is not enforced when generating gcode.  I set feed rates in print settings that are much higher than specified in the machine limits (60x too low) and yet they were honored in the gcode.

I think the labels may have changed, but the interpretation was still mm/s internally to the slicer code that checks limits.  Which really begs the question of why change the labels?  I would rather have consistent settings inside the slicer and it can do the conversion if I ask it to emit for firmware that wants a different unit.  It is easier for me to think in SI units consistently, i.e. mm/s.   Just a thought.

supermerill, 2021-06-10:
I takes care to convert the values where they are used. The label should be good.
Takes care of the top combobox, that decide what to do with these settings. I don't have your project, so i can only guess.

jltx1, 2021-06-10:
[square.gcode.zip](https://github.com/supermerill/SuperSlicer/files/6631663/square.gcode.zip)

Here you can see the line: G1 X104.240 Y100.833 F18000.000
which is mm/min

the settings at the end show: 
; machine_max_feedrate_x = 500
; machine_max_feedrate_y = 500
which are also mm/min for reprapfirmware.  Am I not understanding what is supposed to happen?  I was expecting these to override what I asked for in print settings.

supermerill, 2021-06-10:
`machine_max_feedrate_x` isn't used for safeguarding for now, whatever the unit is.