supermerill/SuperSlicer #1290: Top infill perimiter ironing

TheBoojah, 2021-06-10:
When the angle of rectilinear infill is very shallow (close to parallel) to the perimeters, there is a risk of gaps. See the orange circle below.
In the green areas where the angle is perpendicular there is no problem of course, as here it is easily tuned with infill/perimeters overlap. However if you want no holes in the angled areas, there is a risk of over-extrusion in thin/small areas or elsewhere if the overlap is too high. This has been a problem for me even when ironing, as the small ironing lines does not easily attach to the perimeters if these gaps are too large.
[image]
The filled infill types are a great step to overcome this issue, but it is not feasible to gap-fill these small areas.

Proposal:
An "ironing" perimeter is added. Where the flow is proportional to the angle difference from perpendicular of the infill, perhaps averaged out over a short distance. The flow is 0 at perpendicular and some maximum when the angle is very shallow yet does not trigger gap fill. The location could be 50% of width, meaning right between the inside perimeter and the infill.
This could give a cleaner look, no gaps and less reliance on the overlap parameter and a better base for ironing the whole top layer.

Additional thoughts:
Could be merged with gap fill, i.e. do the gap fill when you reach those areas on the ironing perimeter route. 
The overlap ratio could essentially be removed/ignored, or even be negative and do the ironing perimeter with some minimum flow >0, just like regular ironing.

Thanks for your consideration.

supermerill, 2021-06-10:
some years ago, i modified it to extrude only the first bit of the(missing) link. But prusa rewrote the rectilinear code 3 times since. I didn't have the will to relearn how the thousand(s) lines algorithm works now.
I won't update until it's stable.
I may be able to get the old one, if you want to test it.

There is also some yellow are in the top of your picture, but you don't talk about it. Btw, how your test prints of this look like?

TheBoojah, 2021-06-11:
Here is the actual print:
[image]
So yes there are gaps on the top side too. It may look like it is under-extruded, but the ridges actually protrude so its pretty spot on or even slightly over-extruded.

Wow, yeah that sounds like a mess.
In theory this metod only needs to know the fill_angle, not the details of the actual infill lines. As you are only filling in the holes on average not trying to hit the actual gaps. The input of the function is the angle of the perimeter in world coordinates and the fill angle of the layer.

Pizzahd88, 2021-09-07:
I believe there is a solution that could help a lot on this issue, while resulting in better ironing edges when using Top Infill: Ironing.

If we went around the perimeter with ironing just before doing the top infill Ironing, we could smooth out the bulges normally present around the perimeter and help hide the gaps shown above when infill is connected to perimeter with a shallow angle.
It might not have to go further out than the current infill ironing, staying within the same area as the current top infill ironing implementation.

Shown here with blue, the very first ironing swoop would traverse along the outer bounds of the infill ironing smoothing out any bulges or holes that might be present due to infill<>perimeter angles and over extrusion/flow irregularities.
[image]

It might be that 1 perimeter swoop is enough, maybe we should use 2 to ensure the width of the swoop is covering the entire hole/bulge. 
This implementation should be independent of the infill implementation of PrusaSlicer, and should present better top edges.