supermerill/SuperSlicer #1291: tool tip request

jltx1, 2021-06-10:
### Version
2.3.56

### Operating system type + version
mac0s

### Behavior
enhancement request:
maybe this is a preference setting.

Can you change the tooltips to stay posted until the mouse moves and not time out?  I can't read that fast, especially when I'm trying to think about implications, etc.

supermerill, 2021-06-10:
Not possible on macos, the sdk don't allow it.
Already at max value for all other os.

jltx1, 2021-06-10:
understood.  looks like tool tips are scattered throughout src.  I was hoping localization would push them to one file and I could read there.

supermerill, 2021-06-10:
you can search for them in the resources/localization/slic3r.pot
most of the tooltips (the ones for the settings) are defined in src/libslic3r/printconfig.cpp

Sirage48, 2021-06-11:
jolt1 might like this - not quite perfect but probably nearly everything you need.
[SS-tooltips-1.txt.zip](https://github.com/supermerill/SuperSlicer/files/6636870/SS-tooltips-1.txt.zip)

jltx1, 2021-06-11:
> jolt1 might like this - not quite perfect but probably nearly everything you need.
> [SS-tooltips-1.txt.zip](https://github.com/supermerill/SuperSlicer/files/6636870/SS-tooltips-1.txt.zip)

That's awesome!  Thanks.

BTW, great features you have added.  I have use slic3r for years and even hacked on slic3r PE to try to add some of these features, e.g. gap fill threshold.   Having is superset of features on top of PrusaSlicer gives me the best of both worlds.  Keep up the good work.

closed 2021-06-14 by supermerill