supermerill/SuperSlicer #1292: Error when trying to edit the Filament selection in Configuration Wizard

Duglum, 2021-06-10:
[image]

Duglum, 2021-06-10:
This is in v2.3.56

Sp4rkR4t, 2021-06-11:
Just to add a note to this, if I went through the configuration wizard with default filament profiles to complete the wizard then I get a similar error message whenever I select a printer profile from the dropdown list on the main screen.

[image]

supermerill, 2021-06-11:
> Just to add a note to this, if I went through the configuration wizard with default filament profiles to complete the wizard then I get a similar error message whenever I select a printer profile from the dropdown list on the main screen.
> 
> [image]

You should have created a new issue for that, this will pollute this already-fixed issue.
If I understand, this error happens in a printer "swap", even if you don't touch to the wizard?
If so, can you send me your setting directory to help me reproduce that?

closed 2021-06-26 by supermerill