supermerill/SuperSlicer #1294: Gcode Flavor Change to RepRapFirmware crash

gorf25, 2021-06-10:
### Version
_Version of SuperSlicer used goes here (2.3.56)

### Operating system type + version
_What OS are you using, and state any version Windows 10 latest version

### Behavior
Every time i try to switch the G code Flavor to RepRapFirmware
I get an  unhandled unknown exception, terminating the applaction
Then i get a internal error. invalid map<K,T> Key
* _Steps needed to reproduce the problem
Just try to switch to RepRapFirmware G Code Flavor
* _Expected Results
Switch to RepRapFirmware
* _Actual Results
Crash and program is closed after the 2 dialog boxs
  * _Screenshots if useful_

_Is this a new feature request? No
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
No Project loaded
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

vladimir-poleh, 2021-06-11:
Duplicate of the #1293. Already fixed, just download build 2.3.56.4 again.

gorf25, 2021-06-11:
Sorry but that is the version i am running left off the .4 at the end just downloaded it today..

vladimir-poleh, 2021-06-11:
It was reuploaded few hours ago.

gorf25, 2021-06-11:
Ok Thanks i just re downloaded and now its version  .4b and that is working

Thanks

liftbag, 2021-06-11:
On my mac, when I choose Sprinter as the g-code flavor, RepRapFirmmware appears. Nightly #324.

supermerill, 2021-06-11:
should be fixed in latest nightly

liftbag, 2021-06-14:
> should be fixed in latest nightly

Yes, it's fixed.

closed 2021-06-26 by supermerill