supermerill/SuperSlicer #1299: Volumetric autospeed does not adjust speed to % value

SirStigII, 2021-06-11:
I think I may have noticed a related issue in the latest stable release (2.3.56.4b), if speeds are set to auto like in the above situation, but the outer and top layers are set to a percentage (I was using 50%) they will slice at the max print speed, instead of using the correct percentage. setting the perimeter speed and solid speed to 120mm/s in this case will correctly use the 50% value (60mm/s)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to speed tab
2. set perimeter or solid speed to 0, with autospeed paramaters defined (120mm/s, 12mm^3/s in this case)
3. slice object and change view to "speed"
4. See error

**>> Project File <<**
[problem.zip](https://github.com/supermerill/SuperSlicer/files/6640483/problem.zip)

**Expected behavior**
When external perimeter set to 50% of perimeter autospeed, I expect it to output 50%. (in this case, 60mm/s, 50% of 120mm/s)

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Win 10
 - Version 2.3.56.4b
 - 
**Additional context**
changing the perimeter or solid speed to the discreet value that autospeed is using (120, instead of 0) will correctly change the top or exterior speeds the the 50% value.

supermerill, 2021-06-18:
Changed.
But as it's a big change in the algo, will be for the next beta version (will not be integrated in 2.3.56 branch)

supermerill, 2021-08-09:
should be in this night nightly.

lukeslaboratory, 2021-09-10:
Not sure if related, but if i use 2.3.57 beta, my autospeed is broken and locks at a certain low speed, but using the *exact same* profiles in an earlier version, autospeed works correctly. This happens across filaments and across machine profiles, so it appears to be the slicer itself. I've successfully been using autospeed to 45mm3/s and its correctly worked in the past. 
[image]
Identical settings across the board, but significantly reduced speeds on the newer version

supermerill, 2021-09-28:
@lukeslaboratory like #1563 ?