supermerill/SuperSlicer #1302: Post-process UV cure

DocRuzzy, 2021-06-11:
I would have use for a post-process UV cure similar to the milling post-processing option. This for using UV curable resins extruded via a syringe which beyond my specific interest is highly applicable to bioprinting.

The basic process is:

After every layer switch the UV cure head and pass back and forth (like a 100% rectilinear fill) over an area defined by the skirt.
At a minimum, the UV cure head will need an offset and a spot diameter defined.
The distance between lines will be related to the spot diameter and the travel speed is crucial to the curing.

More complicated options include controlling overlap and intensity.
This is the hardware I'm using and related code.
http://hyrel3d.net/wiki/index.php/UV_and_Clench#Advanced_-_Between_Layers_2

supermerill, 2021-06-22:
I can add a field for start & end gcode, a field to choose the size of the beam, and will do a rectilinear fil over the piece, with '50% overhang' to be sure to go over everything.

DocRuzzy, 2021-06-23:
That sounds great.
Two notes.
1. The start & end gcode would need to be per-layer, the UV cure sweep has to be done at the end of every layer.
2. The 50% overlap is good. The skirt offset distance will need to be set >= the beam size to ensure all printed parts get the double exposure induced via 50% overlap, (shouldn't be a problem, but could be advised in the mouse-over notes).

supermerill, 2021-06-23:
1- of course
2- tooltip can be enough. I don't like to have too many coded safeguards, even more for an exotic feature.

Little note: I have many things to improve or do that many more people will use. So the implementation of this feature may come only in a very distant future. If you need it quickly, your best bet is to implement it yourself (or to hire someone to do the job), I can help with how, where and what to do if needed. It's something like a full day of work for someone that knows the codebase.

DocRuzzy, 2021-06-23:
2. absolutely, I think the tool-tip is fine.

Thanks and I understand. I may have someone to help with this and I'm going to reach out to them.