supermerill/SuperSlicer #1305: Trouble Recovering After "ironing" Top Layer

CCS86, 2021-06-12:
On my otherwise well-tuned bowden printer, I am really struggling with severe under-extrusion, immediately following a section of ironing. I first noticed this when trying the built-in flow calibration test. It was so bad that I couldn't get bed adhesion on the second printed part (not helped by the fact that the test starts with the -20% part).

I made a simpler test object, with a height range modifier, to print a few layers, then iron half that surface, and print one last layer on the other half. After trying a number of different "ironing flow distribution" settings, all the way to 30%, and seeing no real improvement; I finally realized that it is just moving flow from the layer below, up to the ironing pass, for a net-zero flow change.

Is there any way to simply increase the total flow of the ironing pass, without reducing flow underneath?

[image]

CCS86, 2021-06-12:
I now see the "ironing post-process" option. Maybe that is what I need to focus on.

CCS86, 2021-06-13:
I'm having much better luck with ironing PP.

I did a slew of tests, varying ironing flow rate from 10% to 56%, in 2% increments.

Here is a photo of 10% and 56% side by side. There is a shocking lack of difference in the ironed surface, and a definite improvement in the subsequent print section. I'm just surprised at how high I have the flow rate cranked up and it is still a bit under-extruded:

[image]

Here's the STL, if anyone else wants to use it. I use 0.2mm layers, and a 0.4-0.6mm modifier to enable ironing PP and set the flow rate:
[Ironing - Copy.stl.txt](https://github.com/supermerill/SuperSlicer/files/6643198/Ironing.-.Copy.stl.txt)

CCS86, 2021-06-13:
I wish there was an "ironing PP inset" setting. I feel like the ironing path is too close to the outer wall, especially at the start and end:

[image]

supermerill, 2021-06-13:
Do you have linear advance? 
for what I can imagine, the slow-extrusion ironing can "consume" the built pressure in the nozzle, a bit like a "cura coasting", and when you need to print again with a normal flow, it needs to build up pressure again.

There isn't a pressure model yet in susi.

CCS86, 2021-06-13:
I have tried like hell to get PA working well in RRF, but have not had success. I think the lack of a smoothing algorithm really makes it struggle with bowden values and real world prints.

I think you could be right. But, I should be able to test the theory. I did these last tests all at the same speed, 45 mm/s, even the ironing. But, obviously ironing at 45 mm/s is a far lower nozzle pressure.

The thing is, if it was purely a transient response thing, I would expect a faster recovery. The fact that all 3 perimeters and the entire infill is still under-extruded makes me think it is something else.

I'll try printing that final layer at a very low speed, to test the theory.