supermerill/SuperSlicer #1306: Mac OS SS 2.3.56.4 and 5 When minimizing the SS window to the dock, it won't come back up when selecting SS

darrenph1, 2021-06-13:
**Describe the bug**
If I open SS and minimize it to the dock, I can't get it back up unless I quit.

**To Reproduce**
Steps to reproduce the behavior:
1. Open SS
2. Minimize the window to the dock
3. Click on the SS app in the dock
4. Nothing happens

**>> Project File <<**
Not needed

**Expected behavior**
SS window should reappear when called.

**Desktop (please complete the following information):**
 - OS: 2 different systems 10.14.6 same result
 - Version 2.3.5.4 and 5

Sirage48, 2021-06-13:
A temporary work-around.
If you right click on the icon in the dock and click on "Show all Windows" an icon of the SS window appears on the bottom of the screen, then click on that icon and SS returns

supermerill, 2021-06-13:
And this doesn't happen with the 2.3.56.3 or previous versions?

Sirage48, 2021-06-13:
I don't minimise SS. 
Just checked all the versions I have back to 56.0 and this bug is in all of them.
Downloaded 55.5 Super brim ears but it crashes on opening - just flashes the benchy.
[SS-crash.txt.zip](https://github.com/supermerill/SuperSlicer/files/6643662/SS-crash.txt.zip)

Sirage48, 2021-06-13:
I also have an older 2012 iMac - the problem is there also.
I downloaded  and tried 55.5 on the old iMac but it crashed there  BUT it opened the terminal before it crashed.
Both Macs running Big Sur
I have no older SS versions

darrenph1, 2021-06-14:
The bug exist in PS as well so not related to SS.

closed 2021-06-14 by darrenph1