supermerill/SuperSlicer #1310: Excessive amount of gap fill segment is generated

qwewer0, 2021-06-13:
**Describe the bug**
Excessive amount of gap fill segment is generated on 2.3.55.5 and above (latest 2.3.56.5 SS release).

**2.3.56.5**:
- Without gap fill, 51k gcode line is generated
- With gap fill, 174k gcode line is generated (341%)

**2.3.55.5**:
- Without gap fill, 51k gcode line is generated
- With gap fill, 174k gcode line is generated (341%)
- 
**2.3.55.1**:
- Without gap fill, 39k gcode line is generated
- With gap fill, 56k gcode line is generated (144%)

(Wasn't able to test it on 2.3.55.2, as it crashes on generating gcode from stl, and can't load the 3mf file)

**Project File**
[Test.zip](https://github.com/supermerill/SuperSlicer/files/6643907/Test.zip)

**Expected behavior**
The 2.3.55.1 gap fill quantity looks good, but in more details: on the provided project file, the gap fill quantity on a given layer should be close to as mush as one perimeter.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 2.3.56.5

ManuelMcLure, 2021-06-13:
Is this a duplicate of #1206?

qwewer0, 2021-06-13:
> Is this a duplicate of #1206?

Yes, I think it is.
Thanks for reminding me to it.

closed 2021-06-13 by qwewer0