supermerill/SuperSlicer #1311: End of first layer > referring to Q&A #1309

balloni55, 2021-06-13:
Version 2.3.56
description and picture/video see Q&A #1309
the behavior is the same for each of my projects always at the end of first layer
[Platte.zip](https://github.com/supermerill/SuperSlicer/files/6644355/Platte.zip)

supermerill, 2021-06-26:
I don't understand, you linked an issue about support interface and there is no support on your project.

balloni55, 2021-06-26:
Sorry but somebody has deleted my post in Q&A ....
you answered:
Re: [supermerill/SuperSlicer] what happens at the end of the first layer ? (#1309)
Please open an issue with your project file and your gcode.

once again:
At the end of each first layer the nozzel lower in Z-axis and run a triangele.
This deeping is filled by second layer, so it is not a big problem.
[image]
[image]