supermerill/SuperSlicer #1312: Random seam position is not working

Miksu80, 2021-06-14:
In at least few latest versions, the random seam is not working as it should:
Random seam on:
[image]

Aligned seam:
[image]

SuperSlicer
Version: 2.3.56.5
Build: SuperSlicer_2.3.56.5
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19043), 64-bit edition

supermerill, 2021-06-14:
can't reproduce
[image]

but maybe is you share your project file?

Miksu80, 2021-06-14:
I've had a few "issues" that could not be replicated. My config is based on a older prusaslicer -I have modified the config and saved it but the base is from 3-4 years back but could it be the reason for these weird symptoms? Ive attached the bundle and project to zip:

[Piston.zip](https://github.com/supermerill/SuperSlicer/files/6646381/Piston.zip)

supermerill, 2021-06-14:
You're using one-loop perimeters. So the seam is placed where it's possible (where the gap between the outer perimeter and the inner one is the smallest). I didn't add much more options than "near" and "rear" for this one. Now that prusa has moved the seam selection in a new class, maybe i can reuse it a bit to add a bit more options...

Miksu80, 2021-06-14:
Ok, didnt see their connection to each-other, thanks.

closed 2021-06-14 by Miksu80