supermerill/SuperSlicer #1313: Crash when Slicing Complex Objects

BlueDragonX, 2021-06-14:
**Describe the bug**
When loading more complex objects SuperSlicer crashes. Sometimes, if you're patient, loading SS and slicing the same files will work but this is rare.

I first noticed this months ago when slicing Voron 2.4 parts on a much older version. I thought this was an issue with my particular slicer settings but I get the same issue when slicing with the built-in Prusa i3 profiles.

**To Reproduce**
Steps to reproduce the behavior:
1. Load something a bit more complicated (I'm using the [AB-BN-30](https://github.com/VoronDesign/VoronUsers/tree/master/printer_mods/Badnoob/AB-BN) mod for Voron Afterburner).
2. Slice it.
3. Cry.

**>> Project File <<**
[ab-bn-30_Fan.zip](https://github.com/supermerill/SuperSlicer/files/6645934/ab-bn-30_Fan.zip)

**Expected behavior**
More slicey, less crashy.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version: 2.3.56.5 (occurs on previous versions as well).

**Additional context**
I tried using superslicer_console.exe and enabling trace logging. It's not particularly verbose but does confirm that the crash occurs while slicing.

BlueDragonX, 2021-06-14:
Also of note: It simply hangs during slicing just a little bit more frequently than it finishes successfully.

And I did eventually get the attached project file to slice.

supermerill, 2021-06-14:
Can't makes it crash or hang, even on the 7th time.
Tried on 2.3.56.4 and my dev build, on windows 10 x64.

I don't know what's different between our computer to have that different behaviour.
Can you share your setting directory (zipped) ? it's at c:/users/myname/appdata/roaming/superslicer I shouldn't create any difference, but i'm a bit in lack of ideas.

qwewer0, 2021-06-14:
I also were able to slice it.

[image]

BlueDragonX, 2021-06-14:
I was afraid it'd be Schrodinger's bug.

[SuperSlicer-config.zip](https://github.com/supermerill/SuperSlicer/files/6649899/SuperSlicer-config.zip)

I'm probably going to set up my own dev environment and see if I can't figure out what's up. A stack trace would be really handy right about now.

BlueDragonX, 2021-06-15:
Before I get too deep I'll say that my PC was overclocked and that was causing SS to crash. Reducing the clock back to stock no longer has it crashing; instead it consistently hangs now when slicing these more complex parts with the settings I attached.

After building SS locally and inserting an excessive amount of logging I was able to narrow it down to the gapfill.

For my test I've been slicing the [AB-BN-30 duct](https://github.com/VoronDesign/VoronUsers/blob/master/printer_mods/Badnoob/AB-BN/STL_files/AB_BN_30_Hotend-Dragon-front.stl) which is a real hard nut to crack. I found that SS would hang at layer 53 every time. I narrowed this down to the MedialAxis call when generating gapfill extrusions. This particular layer has some thin walls where only two perimeters and a big fat gapfill would fit.

Disabling gapfill allowed the part to slice cleanly but of course removed my gapfill.

Increasing my perimeter widths also allowed the part to slice cleanly. I was running 0.38 and increased it to 0.4.

I'm leaving it set to 0.4 now and have not crashed since. I've also enabled perimeter overlap which reduces the amount of gapfill. So far so good. I still don't know why it hangs on the MedialAxis call other than "my computer is wonky". In any case I'm content with my workaround.

senseisimple, 2021-07-13:
I can confirm that this also crashes SuperSlicer 2.3.56.7 making it likely related to https://github.com/supermerill/SuperSlicer/issues/1389 and https://github.com/prusa3d/PrusaSlicer/issues/6677
I have attached a set of objects that exhibit the crash within a 1mm section there.

Similarly to mentioned here, the crashing segment gomes down to 2 perimeters with gap fill, but i'm not sure why on the specific layer;
SLICES OK:
[image]
CRASHES (object on the right, next layer, z+0.2mm) 
[image]

supermerill, 2021-07-13:
I fixed an old bug in the gapfill in the latest nightly.
I recommend to do your test on it, to be sure it's not already fixed.

senseisimple, 2021-07-13:
@supermerill I've compiled the nightly, issue remains under the same conditions as previously listed. As before, slicing seems to complete but crashes on parsing the gcode/rendering the sliced output. It's pretty clear that there is something going on with the gapfill + perimeter width on an AMD gpu as the trace ends with the invocation of the radeonsi_dri.so

dmesg output
```
[  742.191831] slic3r_main[90525]: segfault at 558b25ca6938 ip 00007fba968f8273 sp 00007ffc5a1a0280 error 4 in radeonsi_dri.so[7fba96707000+fda000]
[  742.191842] Code: 0f 1f 80 00 00 00 00 41 54 55 53 8b 87 50 02 01 00 48 89 fb 48 8b af 48 02 01 00 48 69 c0 18 20 00 00 48 8d bc 07 88 01 00 00 <8b> 07 85 c0 75 17 8b 83 58 02 01 00 85 c0 75 1d 5b 5d 41 5c c3 0f
[  742.191866] audit: type=1701 audit(1626170546.268:241): auid=1000 uid=1000 gid=1000 ses=3 pid=90525 comm="slic3r_main" exe="/usr/share/SuperSlicer/superslicer" sig=11 res=1
```

log output (5:trace)  (segfault):
```
[2021-07-13 05:02:21.292901] [0x00007fbac0f3a080] [info]    Undo / Redo snapshot taken: Import Object: Mashroom (20)-25.49mm-top-2.31mm-broken.obj, Undo / Redo stack memory: 0MB Resident memory: 233MB; Shared memory: 148MB; Private memory: 85MB; Peak memory usage: 233MB
[2021-07-13 05:02:21.344913] [0x00007fbac0f3a080] [debug]   TriangleMesh::repair() started
[2021-07-13 05:02:21.345464] [0x00007fbac0f3a080] [debug]   TriangleMesh::repair() finished
[2021-07-13 05:02:21.345480] [0x00007fbac0f3a080] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2021-07-13 05:02:21.345489] [0x00007fbac0f3a080] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2021-07-13 05:02:21.345542] [0x00007fbac0f3a080] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2021-07-13 05:02:21.346771] [0x00007fbac0f3a080] [debug]   TriangleMesh::repair() started
[2021-07-13 05:02:21.346991] [0x00007fbac0f3a080] [debug]   TriangleMesh::repair() finished
[2021-07-13 05:02:21.347000] [0x00007fbac0f3a080] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2021-07-13 05:02:21.347007] [0x00007fbac0f3a080] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2021-07-13 05:02:21.347027] [0x00007fbac0f3a080] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2021-07-13 05:02:26.167158] [0x00007fba427fc640] [info]    Starting the slicing process. Resident memory: 234MB; Shared memory: 149MB; Private memory: 85MB; Peak memory usage: 234MB
[2021-07-13 05:02:26.167221] [0x00007fba427fc640] [info]    Slicing objects... Resident memory: 234MB; Shared memory: 149MB; Private memory: 85MB; Peak memory usage: 234MB
[2021-07-13 05:02:26.167242] [0x00007fba427fc640] [debug]   Slicing objects - region 0
[2021-07-13 05:02:26.167873] [0x00007fba427fc640] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2021-07-13 05:02:26.167886] [0x00007fba427fc640] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2021-07-13 05:02:26.168103] [0x00007fba427fc640] [debug]   TriangleMeshSlicer::slice
[2021-07-13 05:02:26.168113] [0x00007fba427fc640] [debug]   TriangleMeshSlicer::_slice_do
[2021-07-13 05:02:26.168761] [0x00007fba427fc640] [debug]   TriangleMeshSlicer::_make_loops_do
[2021-07-13 05:02:26.168894] [0x00007fba427fc640] [debug]   TriangleMeshSlicer::slice finished
[2021-07-13 05:02:26.168949] [0x00007fba427fc640] [debug]   TriangleMeshSlicer::make_expolygons in parallel - start
[2021-07-13 05:02:26.169647] [0x00007fba427fc640] [debug]   TriangleMeshSlicer::make_expolygons in parallel - end
[2021-07-13 05:02:26.169661] [0x00007fba427fc640] [debug]   Slicing objects - append slices 0 start
[2021-07-13 05:02:26.169677] [0x00007fba427fc640] [debug]   Slicing objects - append slices 0 end
[2021-07-13 05:02:26.169683] [0x00007fba427fc640] [debug]   Slicing objects - removing top empty layers
[2021-07-13 05:02:26.169690] [0x00007fba427fc640] [debug]   Slicing objects - make_slices in parallel - begin
[2021-07-13 05:02:26.169731] [0x00007fba427fc640] [debug]   Slicing objects - make_slices in parallel - end
[2021-07-13 05:02:26.169747] [0x00007fba427fc640] [debug]   Slicing objects - fixing slicing errors in parallel - begin
[2021-07-13 05:02:26.169758] [0x00007fba427fc640] [debug]   Slicing objects - fixing slicing errors in parallel - end
[2021-07-13 05:02:26.169769] [0x00007fba427fc640] [debug]   Slicing objects - siplifying slices in parallel - begin
[2021-07-13 05:02:26.170056] [0x00007fba427fc640] [debug]   Slicing objects - siplifying slices in parallel - end
[2021-07-13 05:02:26.170128] [0x00007fba427fc640] [info]    Generating perimeters... Resident memory: 235MB; Shared memory: 149MB; Private memory: 86MB; Peak memory usage: 235MB
[2021-07-13 05:02:26.170145] [0x00007fba427fc640] [debug]   Generating perimeters in parallel - start
[2021-07-13 05:02:26.170162] [0x00007fba427fc640] [trace]   Generating perimeters for layer 0
[2021-07-13 05:02:26.170180] [0x00007fba4cda8640] [trace]   Generating perimeters for layer 3
[2021-07-13 05:02:26.170191] [0x00007fba41bfa640] [trace]   Generating perimeters for layer 1
[2021-07-13 05:02:26.170202] [0x00007fba41ffb640] [trace]   Generating perimeters for layer 5
[2021-07-13 05:02:26.170213] [0x00007fba4cda8640] [trace]   Generating perimeters for layer 3, region 0
[2021-07-13 05:02:26.170221] [0x00007fba413f8640] [trace]   Generating perimeters for layer 9
[... cut ...]
strace: Process 345338 detached
strace: Process 345339 detached
strace: Process 345340 detached
strace: Process 345341 detached
strace: Process 345342 detached
strace: Process 345343 detached
strace: Process 345344 detached
strace: Process 345345 detached
strace: Process 345346 detached
strace: Process 345347 detached
strace: Process 345348 detached
strace: Process 345349 detached
strace: Process 345350 detached
strace: Process 345351 detached
strace: Process 345352 detached
strace: Process 345355 detached
```

senseisimple, 2021-07-13:
Related (also AMD):
https://github.com/prusa3d/PrusaSlicer/issues/5745
https://github.com/prusa3d/PrusaSlicer/issues/6342
https://github.com/supermerill/SuperSlicer/issues/1057 (closed, no response)
https://github.com/prusa3d/PrusaSlicer/issues/6345

As noted in many comments, setting env variable `LIBGL_ALWAYS_SOFTWARE=true` alleviates the segfault.
PrusaSlicer >=2.3.1.* does not resolve.

BlueDragonX, 2021-07-13:
If yours is crashing during rendering then I don't think this is related to the crash that I'm experiencing. Mine happens during slicing, specifically when calculating gapfill. I can consistently get mine to work by disabling gapfill. _Sometimes_ it'll work if I enable/disable overlapping perimeters, but it's hit or miss depending on the model. The crash happens upwards of 90% of the time. I've taken to saving the project and repeatedly opening it and hitting the slice button until it works. It's very frustrating but it doesn't happen on that many prints so I guess I'm supposed to live with it.

senseisimple, 2021-07-14:
Perhaps not exactly like yours, symptomatically, it's about the same with the same exact workarounds being applicable, definitely to do with perimeter dimension and gapfill.

It crashes on your model but _after_ slicing (no hang while slicing) when it goes to actually color the features (or whatever the preview mode is set to). 

Changing settings to those as you described alleviates the issue altogether but is suboptimal. Disabling OpenGL hardware rendering fixes it completely but for complex models like yours does slow it down and is a bit more cpu intensive - most smaller/simpler models do just fine.

You didn't provide system information like GPU or actual debugging/trace output or more information on how you arrived at the "MedialAxis" call being the culprit so it'd be difficult to say with certainty if they are related but symptomatically and as for workaround, they are a match. Perhaps Linux does things differently than the Windows version which may be why the trigger is different, even if the end result is the same.

BlueDragonX, 2021-07-14:
I set LIBGL_ALWAYS_SOFTWARE=true as you suggested but that had no effect on my crash. Perhaps it doesn't do anything as I'm in Windows, not sure.

I narrowed it down to MedialAxis the old fashioned way - loads of printf's until I got to the problematic call. I didn't dig further into MedialAxis itself. Again, I'm in Windows, and not really familiar with the debug tools available to me so I fell back to the old standby.

supermerill, 2021-07-16:
@BlueDragonX did you tried the latest nigthly? (available in the actions menu)

BlueDragonX, 2021-07-17:
I just did. It still crashes.

supermerill, 2021-07-20:
> Again, I'm in Windows, and not really familiar with the debug tools available to me so I fell back to the old standby.

Install visual studio and launch the project in debug mode, it should stop where the exception is triggered. (be sure that inside the exception windows, you have the c++ exception checked)

BlueDragonX, 2021-08-10:
It seems like this was fixed with https://github.com/supermerill/SuperSlicer/commit/e9cd1541fa9f72f0a6966674dbe67e10fd5f83be. How was this not causing that many other people issues?

BlueDragonX, 2021-08-10:
I take it back - it works on _most_ files now, but there are still some things where I get consistent crashing. I'll see if I can't drill down into what.

supermerill, 2021-08-11:
Just after alunching the slicer, if you set in the task manager the thread affinity to just one core, does it continue to crash?
If yes, i can try again on my computer.

BlueDragonX, 2021-08-12:
I can't reproduce the crash at all currently. Not sure why that is. I guess it's good...by accident. But this is worrying. I hate heisenbugs.

closed 2021-12-11 by supermerill