supermerill/SuperSlicer #1316: Force Retract when Crossing Perimeter

CCS86, 2021-06-15:
We seem to only have a setting for "only retract when crossing perimeters", but none for "force retract when crossing perimeters".

So, if "only retract when crossing perimeters" is disabled, and you want to allow say 8mm un-retracted travels, you have no way to ensure they are contained within the infill to hide the ooze. They can cross perimeters and cause bad stringing:
 
[image]

CCS86, 2021-12-28:
Hey Merill, any plans for this request?

 I'm finding it hard to balance allowing enough "minimum travel after retraction" to avoid unnecessary retracts, with the behavior it allows: crossing external perimeters without retraction.

supermerill, 2022-01-26:
I just modified a bit this section, i'll add it shortly, unless I forgot about it again.