supermerill/SuperSlicer #1317: Only Custom Start G-code Prevents Filament Specific GCode

randellhodges, 2021-06-16:
If I check the box on the Printer Settings->Custom G-Code for Only custom Start G-code, the value in Filament Settings->Custom G-code->Start G-code is not emitted.

Superslicer version: 2.3.56.5

randellhodges, 2021-06-24:
This might be a case of, "be careful what you ask for", but I tried the nightly and it seems to be working but...

[image]

Sorry, my picture is backwards. The one on the right does not have "only custom start g-code" selected, the one of the left has that option selected.

The lines in yellow are the ones I wanted gone, and the ones in red should probably be down where the yellow lines are currently.

I guess I could include the G21, G90, and M83 in my PRINT_START macro, but if I ever wanted to toggle relative/absolute extrusion in the slicer, I'd have to update the klipper macro. I am checking now to see if there is a way I can pass that choice to my macro.

supermerill, 2021-06-24:
write `{if use_relative_e_distances}G91{else}G90{endif}`

randellhodges, 2021-06-24:
Yup, I'll close it. I just wanted to "document" it so maybe it'll help someone else out in the future. Thanks!

closed 2021-06-24 by randellhodges