supermerill/SuperSlicer #1320: Feature Request: Perimeter Z Extension (Strength Reinforcement)

dingyifei, 2021-06-17:
**Is your feature request related to a problem? Please describe.**

When I'm printing objects with axels or long thin sticks with one end attached to another shape, I find the layer below the layer two things are joint together is the weak point. Below is the picture of a failed print. 

Currently, this kind of intersection is being treated like below when a infill below 100% is used:

[image]
[image]
[image]

**Describe the solution you'd like**

When a smaller perimeter joins another larger perimeter, extend the smaller perimeter into the larger perimeter $x$ layers. It allows the perimeter to join with more infill material and potentially provides better stress relief. The solid infill below area doesn't solve the problem since turning it up drastically increased the filament used.

**Describe how it would work**

It works when one small perimeter is above/below a larger perimeter when the smaller perimeter is x% smaller than the one below it. Where on the graph, the left 2D illustration shows the current slicing result, and the right one shows the solution I'm proposing. I think it could also be a half-sphere projecting downward, so the perimeter gets smaller when it gets more into the larger object to reduce the material used.

[image]
[image]

**Describe alternatives you've considered**

Another solution is to add a solid infill above the intersection ratio, where the larger v.s. smaller perimeter size meets $x$% difference, the solid infill is added for $y$ layers of the larger perimeter.

**Additional context**
Add any other context or paint drawing about the feature request here.

dingyifei, 2021-06-17:
I forgot to attach the failed print, below is the picture:
[image]

supermerill, 2021-06-17:
It also happens to me.
But when I tried to replicate it a year after to create a feature to address that, I couldn't snap the axle on that layer anymore.
So I attributed the problem to underextrusion.

I won't do a thing until I can reproduce the problem. If you have a simple & quick to print stl to reproduce it, I can give it a try.

CCS86, 2021-06-17:
As a mechanical engineer, I can't see how changing the print settings many layers below your failure layer could possibly change the outcome.

supermerill, 2021-06-17:
> As a mechanical engineer, I can't see how changing the print settings many layers below your failure layer could possibly change the outcome.

Perimeter-on-perimeter can have better bonding than perimeter-on-top fill if the top fill is a bit under-extruded. uner-extruded perimeters have "holes" on the side, so printing another under-extruded perimeter on top of it creates a good bonding in the middle of the extrusion. the top fill doesn't follow the perimeter path, so there are many possible weak points.

Note that 1-perimeter-on-top mitigate the issue, as almost all of the perimeters arrive one layer early.

CCS86, 2021-06-17:
I can agree with that, but I'd argue the issue there is under-extrusion.

Plus, his part failed between layers with the same pattern, not at the junction layer of the tower.

neophyl, 2021-06-17:
One thing I do in similar situations is to design the part with this phenomenon in mind, I add in thin hollows into the design.  This forces the slicer to add perimeters into those areas in exactly the places I need them.
You could add in a modifier mesh which would do the same thing but thats more difficult to do.  Much easier to do it in cad to start with.  Its all about designing with your production medium in mind.

supermerill, 2021-06-17:
> One thing I do in similar situations is to design the part with this phenomenon in mind, I add in thin hollows into the design. This forces the slicer to add perimeters into those areas in exactly the places I need them.

if it's possible to have the modification directly on the slicer, it's easier as sometimes you can't modify the mesh easily.

neophyl, 2021-06-17:
I print/make a lot of lasertag and airsoft based scratch builds and mods so most of what I print is 'practical' in that they are handled and used a lot.  So far I've not had a lot break (excepting stuff when dropped onto concrete which is to be expected).  So I don't think my printer setup is under extruding.  I also usually make use of the extra perimeters on alternating layers a lot for my stuff just to give it a bit more thickness and durability on larger parts.  That's probably my favorite addition to Susi :)

There is a particular part, its an extension to a printed glock frame magwell.  To fit it you have to sort of snap fit it onto the frame and it stresses the upright 'tower' as its fitted.  Lateral stresses so puts most strain against the layer lines.  In the un-reinforced version I was snapping them about 30-40% of the time.  With the reinforced with a couple of voids version I haven't had one snap yet.

Placing a small modifier completely inside an object is a pain in the rear which is why I opened a feature request over on the Prusa github for a 'xray' / translucent mode so you could see inside stuff to place the modifiers.  I eventually just realised it would be easier to just model them in to start with.

I've attached a 3mf with the part in question with both reinforced and not reinforced versions.  I apologize for the errors in the models.  The original model wasn't mine and I haven't managed to repair all the errors but I managed to reduce them enough so it slices fine.  One day I may just get around to just re-modelling it completely.

I also spend a fair amount of time on the Prusa forum, and I recall others having issues with parts printed on to solid top infill breaking off at that change.  One I remember was this one https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-how-do-i-print-this-printing-help/how-to-internally-strengthen-a-post-going-into-a-platform-in-cad-stl/  

If I can be of any other help just shout out.

Phil
[tower adhesion reinforcement test_susi_2_3_56.zip](https://github.com/supermerill/SuperSlicer/files/6673309/tower.adhesion.reinforcement.test_susi_2_3_56.zip)

dingyifei, 2021-06-18:
I'm gonna print some extrusion calibration first and try to replicate the issue after I can rule out under extrusion.

After I figured out the extrusion I will try to print a simple part trying to replicate the issue, where I'll try to print it in the following ways

1. normal
2. put a tiny gap in the center to add parameters in the connection
3. Put a through hole in the axel

I'm still doing the extrusion tests, will attach the stls later.

dingyifei, 2021-06-18:
I realized 3 is a silly idea... So now it's 
1. Normal with 31% infill
2. Tiny gap in the center
3. Normal with 100% infill

The settings, and STLs I used are attached below in a zip file.
[Downloads.zip](https://github.com/supermerill/SuperSlicer/files/6673852/Downloads.zip)

Here are some images of the sliced parts
[image]
[image]

I'll try to snap them off by hand instead of attaching them to a clamp to hang the parts since I don't have a good clamp at the moment.

dingyifei, 2021-06-18:
I don't think it's caused by under extrusion since even after I over extruded the prints a similar pattern occur.

Indeed the hollow-based axel (1) is weaker than the other two and breaks off on the second layer again, but the other two break off from the topmost layer of the base (where it comes down to layer adhesion after all). I wonder if 2 and 3 are still beneficial in better layer adhesion. Perhaps instead of extending the smaller perimeter, combining two layers of the smaller perimeter and everything inside it into a taller one like the combined infill feature would help?

[image]
[image]

CCS86, 2021-06-18:
Those two solid versions look about as good as it gets. Besides changing part geometry to reduce stress in that area, the only way I can think to increase strength there, is to have a fan speed reduction in the first couple layers at the junction. You could easily do that with a height range modifier.

closed 2021-06-20 by dingyifei