supermerill/SuperSlicer #1323: estimated printtime is not correct

Hopser2k, 2021-06-17:
Hi, since the "Spacing editions" i discovered that the estimated print times are much lower than before.
If I have sliced an object and printed it, The estimated time was nearly perfect.
But since the new Versions, it the estimated printing time is much lower than before.
Superslicer tells me it could be a print time of 5h 30 sec. Octoprint tells me 7:45. And the Print was 7:55
Before the new versions, I have only a difference of 20 or 30 Minutes for such a print.
It is the same object  with the same settings, but different times.
Hope You can understand my english.

Best regards
Uli

supermerill, 2021-06-17:
I don't have your project. can't test. Can you paste it (zipped) here ?

supermerill, 2021-06-18:
please?

Hopser2k, 2021-06-19:
[flexi turtle.zip](https://github.com/supermerill/SuperSlicer/files/6680164/flexi.turtle.zip)
[flexi turtle 3mf.zip](https://github.com/supermerill/SuperSlicer/files/6680166/flexi.turtle.3mf.zip)

Here the missing files.