supermerill/SuperSlicer #1325: problem with the calibration menu tabs

erathos77, 2021-06-17:
**Describe the bug**
When the included calibration objects are generated from any calibration tabs, it shows an inappropriate alert in the 3D view window:

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'any calibration menu tab'
2. Click on "generate" in the help window
3. go to the 3D view window
4. See error: Alert at the bottom of the window
"Error: 
An object outside the print area was detected
Resolve the current problem to continue slicing."

**Desktop (please complete the following information):**
 - OS: macOs 10.13
 - Version [e.g. 2.3.56.5], 

**Additional context**
Despite of this  no such other object is visible in the window, and the calibrations perform well from the gcode.
Maybe intruders in the included calibration stl files ?

kevenwyld, 2021-06-17:
All of the calibrations are created using amf files, not stl.  I have noticed that when opening amf files from FreeCAD where I create most of my models, they are imported into superslicer off of the build plate and I get this error.  Then I have to click `Arrange` from the tool bar to center it.

I had always assumed this was a FreeCAD export issue but I just tried this with one of the amf files from a calibration directory and it does the same thing:
[image]

I wonder if when you click generate it imports the amf file and then positions it so for a brief moment it's outside the bounds of the build plate?

supermerill, 2021-06-17:
yes.

liftbag, 2021-06-18:
Seems solved with the MacOS nightly #364.

supermerill, 2021-06-18:
be careful ot not write the '#', as github will think that you want to link to an issue.

liftbag, 2021-06-18:
> be careful to not write the '#', as github will think that you want to link to an issue.

Understand.

falo2k, 2021-06-29:
I just ran into this in the latest release when trying to drop a pair of Voron cubes for the dimensional accuracy test.

liftbag, 2021-07-01:
> I just ran into this in the latest release when trying to drop a pair of Voron cubes for the dimensional accuracy test.

I can confirm. In my case it happens because the cube is positioned at the end of the axes instead of in the center of the bed.

supermerill, 2021-07-02:
Somone updated the voron calibration cube. I didn't checked that it was correctly at 0,0...
Will change that

closed 2021-08-09 by supermerill