supermerill/SuperSlicer #1326: Feature-Dependent Acceleration should switch to specified accel before (not after) entering that feature

Poofjunior, 2021-06-18:
First off, feature-dependent acceleration is implemented, which is awesome.
[image]
The problem is that the acceleration setting is arriving *late*--after it starts that feature. It would be better if the new setting takes place upon exiting the previous feature.

Here's the current behavior. Let's say perimeter accel is slow, but infill accel is fast. We're printing infill with fast acceleration. Then, the infill feature completes, and the printer travels to the start of the next perimeter *using the infill acceleration value.* The result is that this rapid accel causes more ringing on the part at the start of the next perimeter, which kinda reduces the quality improvements you'd expect from this feature. Instead, when the infill is complete, it should use the perimeter accel setting to travel from the final infill location to the start of the next perimeter.

Generally, when one feature completes, it should immediately switch to the acceleration setting of the *next* feature *before* traveling to the beginning of that next feature.

This is the current behavior on 2.3.56.5: Wild Spacing Edition.

Happy to talk this through in more detail if you want to chat about this!

Thanks a bunch for the fantastic project development!

supermerill, 2021-06-18:
With 800 for perimeter, 1500 for infill and 1000 for all others:

```
G1 F900
M204 P800 ; adjust acceleration
G1 X145.748 Y145.748 E0.42294 ; Internal perimeter
G1 X154.252 Y145.748 E0.42294 ; Internal perimeter
G1 X154.252 Y154.252 E0.42294 ; Internal perimeter
G1 X145.823 Y154.252 E0.41921 ; Internal perimeter
M204 P1000 ; adjust acceleration
G1 X145.262 Y154.738 F10800.000 ; move to first External perimeter point
;TYPE:External perimeter
;WIDTH:0.525
G1 F900
M204 P800 ; adjust acceleration
G1 X145.262 Y145.262 E0.44765 ; External perimeter
G1 X154.738 Y145.262 E0.44765 ; External perimeter
G1 X154.738 Y154.738 E0.44765 ; External perimeter
G1 X145.337 Y154.738 E0.44411 ; External perimeter
M204 P1000 ; adjust acceleration
G1 X145.512 Y154.304 F10800.000 ; move inwards before travel
;WIPE_START
G1 F8640.000
G1 X145.305 Y150.623 E-0.95000 ; wipe and retract
;WIPE_END
G1 E-0.05000 F2100.000 ; retract
G1 X146.825 Y146.023 F10800.000 ; move to first Solid infill point
G1 E1.00000 F2100.000 ; unretract
;TYPE:Solid infill
;WIDTH:0.524374
G1 F900
M204 P1500 ; adjust acceleration
G1 X146.247 Y146.600 E0.03875 ; Solid infill
G1 X146.247 Y147.305 E0.03343 ; Solid infill
G1 X147.305 Y146.247 E0.07098 ; Solid infill
G1 X148.010 Y146.247 E0.03343 ; Solid infill
```

Poofjunior, 2021-06-18:
Yeah; this actually demonstrates it perfectly.
```
G1 F900
M204 P800 ; adjust acceleration
G1 X145.748 Y145.748 E0.42294 ; Internal perimeter
G1 X154.252 Y145.748 E0.42294 ; Internal perimeter
G1 X154.252 Y154.252 E0.42294 ; Internal perimeter
G1 X145.823 Y154.252 E0.41921 ; Internal perimeter
M204 P1000 ; adjust acceleration. ## This should be 800, so that we approach the Ext perimeter with 800 accel, not 1000.
G1 X145.262 Y154.738 F10800.000 ; move to first External perimeter point
;TYPE:External perimeter
;WIDTH:0.525
G1 F900
M204 P800 ; adjust acceleration
G1 X145.262 Y145.262 E0.44765 ; External perimeter
```

In a nutshell, approaching the next feature should use the next feature's acceleration setting. 

So in the above example, the move on line 8
```
G1 X145.262 Y154.738 F10800.000 ; move to first External perimeter point
```
should be using 800 accel. But in the line just above, it is set to the "everywhere else" setting, which is 1000.

CCS86, 2021-06-18:
I like this idea as well, as I have noticed ringing introduced by high travel accel, as the printer moves to start lower accel perimeters. Having a check-box that allowed perimeter accel values to be applied to the perimeter start point travel move would be a great feature.

supermerill, 2021-06-18:
that reminds me that someone requested a travel acceleration control
#435

moving the acceleration change before the move is very easy. But won't it have big drawback? Like slow big travel when you have to jump to other side of your print to extrude that tiny gap fill on this layer?

CCS86, 2021-06-18:
That's the same idea that popped into my head. It would be interesting to see if the firmware can execute that code without a significant motion hiccup. 

At least in RRF, the command would need to be:

`M204 T____`

in order to affect travel moves.

supermerill, 2021-06-18:
What is defined as "travel" for reprapfirmware?
I'm going to write  `M204 S_myaccel T_myaccel` for reprapfirmware.

CCS86, 2021-06-18:
I think the S parameter is deprecated. It uses P and T now:

https://duet3d.dozuki.com/Wiki/Gcode#Section_M204_Set_printing_and_travel_accelerations

I think the distinguishing feature between print/travel moves is the presence of an extrusion move, not G1/G0

Poofjunior, 2021-06-18:
> Maybe if the travel is cut in two chunk, like that:
> 
> ```
> M204 P1000 ; adjust acceleration: default
> G1 X140.000 Y150.000 F10800.000 ; move to first External perimeter point (p1)
> M204 P800 ; adjust acceleration: perimeter
> G1 X145.262 Y154.738 F10800.000 ; move to first External perimeter point (p2)
> G1 X145.748 Y145.748 E0.42294 ; Internal perimeter
> ```
> 
> one for accel and one for decel

This could be one way to handle it! I should mention that, with this solution, we are assuming how long it takes for a printer's ringing to die down before starting the next move. Here, the ringing must die down over the duration of the small "decel" move before the next move begins.

supermerill, 2021-06-18:
In theory, if the firmware planner is well done it should start the last bit of travel with a  low enough velocity for the low accel to stop the extruder.

Currently, if the travel is strait, I switch in the middle.
If it's a perimeter dodge, it will try to get at least 20*nozzle size dist to slow down (or middle it it's too short)

CCS86, 2021-06-18:
Maybe if the travel move is short enough (say less than 3mm), you just apply the accel value before the move.

Poofjunior, 2021-06-19:
> In theory, if the firmware planner is well done it should start the last bit of travel with a low enough velocity for the low accel to stop the extruder.
>
> Currently, if the travel is strait, I [...]

I think this will probably be plenty sufficient. Thanks for looking into making some adjustments.

supermerill, 2021-08-09:
commit 7ae60b6c82e3d6d4448044db6a3298b8f87189f4
should be in this night nigthly