supermerill/SuperSlicer #1327: FR: M1 support

kr15uk, 2021-06-18:
Feature Request: M1 AppleSilicone aka ARM support, PrusaSlicer does that from v2.3.1 so should be piece of cake, maybe?! Thanks.

supermerill, 2021-06-18:
I don' know how to compile for it. I don't have a mac either.

boardthatpowder, 2021-08-09:
On my M1 Superslicer sometimes crashed, but most times it would hang forever when attempting to slice. I followed the instructions from https://github.com/supermerill/SuperSlicer/blob/master/.github/workflows/ccpp_mac.yml to build from source as a workaround, and happy to report all working no issues. The only thing I changed in the instructions was the target osx version.

supermerill, 2021-08-11:
If you want to share your natively compiled release i can add it to the release page.
I hope that github will add m1 vms quickly.

dianlight, 2021-09-07:
@boardthatpowder what version CMAKE_OSX_DEPLOYMENT_TARGET do you use for compiling in m1 native?

hoppelfuss, 2021-09-12:
Just use the latest version 11.5.2, works perfectly fine.

samwiseg0, 2021-11-21:
Here is version 2.3.57.6 compiled for arm MacOS (11.6). 

https://www.dropbox.com/sh/34syqj6e7c20mz6/AAA097uOTzWH7Ih8kqJkgtxWa?dl=0

jallen810, 2021-12-01:
Has anyone tried to build for MacOS 12.x?? I just got a 14" MBP M1Pro. I'm currently running SS with the normal Mac build and it seems fine but I'm curious if there would be advantages to installing the native code? TIA

samwiseg0, 2021-12-01:
> Has anyone tried to build for MacOS 12.x?? I just got a 14" MBP M1Pro. I'm currently running SS with the normal Mac build and it seems fine but I'm curious if there would be advantages to installing the native code? TIA

The ones I built in the comment above yours should work just fine.

radaiko, 2021-12-12:
> @thatguychad
> 
> Run this in terminal:
> 
> > xattr -cr /Applications/SuperSlicer.app

This should be in the release notes for 2.3.57.7.

radaiko, 2021-12-13:
> I tried the builds, but they show as corrupt for me #2055

Look at my comment. You have to run the terminal command when switching from INTEL to APPLE Silicon version.

jallen810, 2021-12-21:
> > I tried the builds, but they show as corrupt for me #2055
> 
> Look at my comment. You have to run the terminal command when switching from INTEL to APPLE Silicon version.

Sorry for the delay, finally got around to downloading the latest Stable (Sanjay Ver) and today's nightly. After running the 'xattr -cr /Applications/SuperSlicer.app` terminal command in terminal as suggested. I am also still getting the "app is corrupt, you should delete it" error. Has anyone made any progress on this front? I read in the #2055 that we might need to have SS certified under an Apple Dev credentials, is that the reason we can't run this? (Sorry not an AppDev) trying to learn a little bit about this to understand the issue

samwiseg0, 2021-12-21:
@jallen810 It does indeed work after running that command. Are you sure you are installing it to you application folder and running that version after you run the command? 

Keep in mind that if the image is still mounted finder will try and launch the one in the image instead of the one in your application folder if you search for it in spotlight.

kr15uk, 2021-12-21:
M1 2020 MBP Monterey clean install, downloaded DMG, app dumped in Applications, xattr command on app, right click open and all good. Make sure you haven't done "keep both" when copied across and xattr is pointing at the app copy you are trying to run, also it's copied locally and not run from DMG.

jallen810, 2021-12-21:
> @jallen810 It does indeed work after running that command. Are you sure you are installing it to you application folder and running that version after you run the command?
>
> Keep in mind that if the [...]

I figured it out. I'm a dumb dumb, I have several versions installed and I post-script the name as "SuperSlicer.ARM12.20.21" so I had to adjust the terminal command to match... now it works fine. Thanks guys

RayPS, 2021-12-23:
Can somebody here from this thread help confirming this #2111 probably M1 only bug? 
The current latest Arm build from release page (2.3.57)

mikhailbot, 2022-01-30:
Attempted to run `./BuildMacOS.sh -ia` on the 23dc9d0 which is what 2.3.57.9 was tagged against but getting the below error, anything to try?

Mac Mini M1 on macOS 12.1

```shell
$ xattr -cr /Applications/SuperSlicer.app

$ open /Applications/SuperSlicer.app     
The application cannot be opened for an unexpected reason, error=Error Domain=NSOSStatusErrorDomain Code=-10827 "kLSNoExecutableErr: The executable is missing" UserInfo={_LSLine=3829, _LSFunction=_LSOpenStuffCallLocal}
```

supermerill, 2022-01-30:
I don't know. I can only debug when building from intel cpu. Someone with an arm one can debug?

samwiseg0, 2022-01-30:
> Attempted to run `./BuildMacOS.sh -ia` on the 23dc9d0 which is what 2.3.57.9 was tagged against but getting the below error, anything to try?
>
>
>
> Mac Mini M1 on macOS 12.1
>
>
>
> ```shell
>
> $ xattr -cr [...]

Not sure what your issue might be but it does build and run fine on arm based macs. 

You can find the a working arm build here https://www.dropbox.com/sh/34syqj6e7c20mz6/AAA097uOTzWH7Ih8kqJkgtxWa?dl=0

mikhailbot, 2022-02-05:
Thanks for that, will try it slowly when I have time--I grabbed the rebuilt one and it runs much smoother and faster!

Is there any CI/CD provider that supports Apple Silicone yet? Or maybe we need to chip in to get @supermerill an M1!

tkodev, 2022-02-11:
> Thanks for that, will try it slowly when I have time--I grabbed the prebuilt one and it runs much smoother and faster!
> 
> Is there any CI/CD provider that supports Apple Silicone yet? Or maybe we need to chip in to get @supermerill an M1!

Apple silicone :p 

I know of several but it is likely better if we see if the existing CI/CD service used (assuming there is one) supports it. 

I worked with Microsoft AppCenter to build iOS applications - possibly a service we can research into

supermerill, 2022-02-22:
I'm testing on `mater_dev`

claytono, 2022-02-22:
I did manage to build this for arm, with no libiconv issues.  I did run into a separate issue, I get an error when linking the final binary saying that `Slicr` is a directory that already exists.  It is true that there is a directory named `slicr` in the build directory, and since MacOS FS is case insensitive, it can't create the binary.  In the short term I've worked around it by having it link to `Slicr.bin`.

Are you building for arm or x86?  I'll try an x86 cross compile in a bit.  I don't have an x86 mac anymore, but probably will in a few days.

supermerill, 2022-02-22:
It's a cross-compile, yes. Using the macos-11 github vm.

claytono, 2022-02-22:
Ok, that makes sense.  I've reproduced it locally doing a cross-compile.  I do remember seeing something like this when I was doing the initial work, so hopefully looking into it will remind me of the previous issue.

claytono, 2022-02-22:
I haven't determined why, but it appears that only the linkflags are being passed into b2 when configuring boost, and the cflags variants are being left out:

```
cd /Users/coneill/dist/SuperSlicer/deps/build/dep_Boost-prefix/src/dep_Boost && ./b2 cflags=-fPIC cxxflags=-fPIC "linkflags=-arch x86_64" -j10 --with-system --with-iostreams --with-filesystem --with-thread --with-log --with-locale --with-regex --with-date_time --layout=versioned --debug-configuration toolset=clang address-model=64 link=static threading=multi boost.locale.icu=off --disable-icu release stage
```

the cflags/cxxflags should have `-arch x86_64` here like the linkflags do.

claytono, 2022-02-22:
The problem is this block: https://github.com/supermerill/SuperSlicer/blob/6ba0a339a6e914e36d1eca288aada952f054ddcc/deps/Boost/Boost.cmake#L111-L120

Because `UNIX` is true on MacOS, the `APPLE` block never gets run.  The fix is to reverse the order of the test like this: 

```diff
@@ -109,14 +109,14 @@ if (IS_CROSS_COMPILE AND APPLE)
 endif ()

 set(_boost_flags "")
-if (UNIX)
-    set(_boost_flags "cflags=-fPIC;cxxflags=-fPIC")
-elseif(APPLE)
+if(APPLE)
     set(_boost_flags
         "cflags=-fPIC ${_arch_flags} -mmacosx-version-min=${DEP_OSX_TARGET};"
         "cxxflags=-fPIC ${_arch_flags} -mmacosx-version-min=${DEP_OSX_TARGET};"
         "mflags=-fPIC ${_arch_flags} -mmacosx-version-min=${DEP_OSX_TARGET};"
         "mmflags=-fPIC ${_arch_flags} -mmacosx-version-min=${DEP_OSX_TARGET}")
+elseif (UNIX)
+    set(_boost_flags "cflags=-fPIC;cxxflags=-fPIC")
 endif()

 set(_build_cmd ${_build_cmd}
```

That gets boost to cross compiling properly, but I'm still seeing other libraries like libjpeg that look to also be failing due to missing arch flags.  I haven't had time to look into that.

sammcj, 2023-04-10:
It looks like the latest builds are still intel only.

- Is there anything I can do to help test builds for this?

sammcj, 2023-04-15:
I had a crack at doing a build for Apple Silicon/ARM but ended up with so many errors I couldn't get it to work 🥲, if anyone has a build of the latest version please share it and I'll do some testing.

sammcj, 2023-04-19:
@thatguychad interesting - I must be doing something wrong then 🤔.

There's a lot of problems but I don't want to hijack this thread - I see this repo has discussions enabled so I might start a thread there although I see someone has done a build and uploaded it already which I must just use - https://github.com/supermerill/SuperSlicer/discussions/3692